CM-Huawei-P8-Development/device_gra_l09 #1: LTE signal level

AD2076, 2016-11-05:
Trying with config_LTE_RSRP_threshold_type=2 in
overlay/frameworks/base/core/res/res/values/config.xml
not a perfect solution but at least we should have (a not very accurate) signal strength

Nexolight, 2016-11-07:
Hm does it work?
As far as I remember the signal wasn't reported correctly by the lower end libraries and thus the framework should be unable to have a correct value. 

But well I never tested it as I noticed later that the overlay wasn't even used when I changed this back then. Feel free to commit it when it displays better values.

SpeedoWBT, 2017-03-20:
Hi Nexolight , it seem that signal can be correct with this patch (4 february ) from Xepeleato
https://github.com/XePeleato/android_device_honor_berlin/commit/9057527c107aa4c892ba8241719a03656e6dcc38

Nexolight, 2017-03-21:
Could be worth a try but I used almost any ril library from huawei to get it working. I just know that I could replace libreference-ril.so and librilutils.so without issues.

Nexolight, 2017-05-18:
AD2076 extended the main class and has overwritten the part responsible for the signal strength. It seems to help but It's not that accurate yet.