CM-Huawei-P8-Development/device_gra_l09 #2: Camera

AD2076, 2016-11-09:
Have a look here: https://github.com/Meticulus/android_device_huawei_hi6250/commit/abd6d8d671da9c6f042e851b3cbed9dff76f1f37

worth a try

Nexolight, 2017-01-19:
Ok started to investigate.
The Camerapp itself fails due to not finding any camera.
I assume This HwCamCfgSrv is required to initialize the camera as it's done in the vendors init.rc files.

For completeness I will note the steps I did just in case it won't succeed at the end and someone wan't to pick up the work.

Copying 
/system/bin/HwCamCfgSvr
/system/lib/libHWCamCfgSvr.so
/system/lib64/libHWCamCfgSvr.so
/system/lib/libHWCamCfgSvr_jni.so
/system/lib64/libHWCamCfgSvr_jni.so

Results in:
```
01-19 13:59:30.693 10290 10290 I ImagingSystem: CImagingSystemDriver::CImagingSystemDriver(char*, char*, bool) enter !
01-19 13:59:30.695 10290 10290 E ImagingSystem: failed to dlopen(/system/lib/hwcam/hwcam.GRACE.m.hi3635.so)!
01-19 13:59:30.695 10290 10290 F ImagingSystem: hw::CNoAddRefRelease4T<T>* hw::CIntfPtrBase<T>::operator->() const [with T = hw::IImagingSystem; hw::CIntfPtrBase<T>::_TPtrNoAddRefRelease = hw::CNoAddRefRelease4T<hw::IImagingSystem>*]: SHOULD NOT BE NULL!
01-19 13:59:30.695 10290 10290 F libc    : Fatal signal 6 (SIGABRT), code -6 in tid 10290 (HwCamCfgSvr)
01-19 13:59:30.696  2925  2925 I DEBUG   : property debug.db.uid not set; NOT waiting for gdb.
01-19 13:59:30.696  2925  2925 I DEBUG   : HINT: adb shell setprop debug.db.uid 100000
01-19 13:59:30.696  2925  2925 I DEBUG   : HINT: adb forward tcp:5039 tcp:5039
01-19 13:59:30.747  2925  2925 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-19 13:59:30.747  2925  2925 F DEBUG   : CM Version: '13.0-20161228-UNOFFICIAL-hwgra'
01-19 13:59:30.747  2925  2925 F DEBUG   : Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/8f0c8d25f6:userdebug/test-keys'
01-19 13:59:30.747  2925  2925 F DEBUG   : Revision: '0'
01-19 13:59:30.747  2925  2925 F DEBUG   : ABI: 'arm'
01-19 13:59:30.747  2925  2925 F DEBUG   : pid: 10290, tid: 10290, name: HwCamCfgSvr  >>> system/bin/HwCamCfgSvr <<<
01-19 13:59:30.747  2925  2925 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
01-19 13:59:30.748  3307  3463 W NativeCrashListener: Couldn't find ProcessRecord for pid 10290
01-19 13:59:30.750  2925  2925 F DEBUG   : Abort message: 'hw::CNoAddRefRelease4T<T>* hw::CIntfPtrBase<T>::operator->() const [with T = hw::IImagingSystem; hw::CIntfPtrBase<T>::_TPtrNoAddRefRelease = hw::CNoAddRefRelease4T<hw::IImagingSystem>*]: SHOULD NOT BE NULL!
01-19 13:59:30.750  2925  2925 F DEBUG   : '
01-19 13:59:30.750  2925  2925 E DEBUG   : AM write failed: Broken pipe
01-19 13:59:30.750  2925  2925 F DEBUG   :     r0 00000000  r1 00002832  r2 00000006  r3 f76a2b8c
01-19 13:59:30.750  2925  2925 F DEBUG   :     r4 f76a2b94  r5 f76a2b44  r6 00000002  r7 0000010c
01-19 13:59:30.750  2925  2925 F DEBUG   :     r8 00000000  r9 00000000  sl 00000000  fp ffd91d3c
01-19 13:59:30.750  2925  2925 F DEBUG   :     ip 00000006  sp ffd91860  lr f756954d  pc f756b93c  cpsr 400f0010
01-19 13:59:30.753  2925  2925 F DEBUG   :
01-19 13:59:30.753  2925  2925 F DEBUG   : backtrace:
01-19 13:59:30.753  2925  2925 F DEBUG   :     #00 pc 0004293c  /system/lib/libc.so (tgkill+12)
01-19 13:59:30.753  2925  2925 F DEBUG   :     #01 pc 00040549  /system/lib/libc.so (pthread_kill+32)
01-19 13:59:30.753  2925  2925 F DEBUG   :     #02 pc 0001c44f  /system/lib/libc.so (raise+10)
01-19 13:59:30.754  2925  2925 F DEBUG   :     #03 pc 00019601  /system/lib/libc.so (__libc_android_abort+34)
01-19 13:59:30.754  2925  2925 F DEBUG   :     #04 pc 00017514  /system/lib/libc.so (abort+4)
01-19 13:59:30.754  2925  2925 F DEBUG   :     #05 pc 0000355b  /system/lib/liblog.so (__android_log_assert+86)
01-19 13:59:30.754  2925  2925 F DEBUG   :     #06 pc 0000ab89  /system/bin/HwCamCfgSvr
01-19 13:59:30.754  2925  2925 F DEBUG   :     #07 pc 000068f5  /system/bin/HwCamCfgSvr
01-19 13:59:30.754  2925  2925 F DEBUG   :     #08 pc 000173c1  /system/lib/libc.so (__libc_init+44)
01-19 13:59:30.754  2925  2925 F DEBUG   :     #09 pc 000069d4  /system/bin/HwCamCfgSvr
01-19 13:59:30.801  2925  2925 F DEBUG   :
01-19 13:59:30.801  2925  2925 F DEBUG   : Tombstone written to: /data/tombstones/tombstone_05
```

Nexolight, 2017-01-19:
These are enough to get our perfhub to run

```
/system/lib/libperfhub_client.so
/system/lib/libperfhub.so 
/system/lib/libperfhub_service.so 
/system/bin/perfhub 
/system/etc/permissions/com.hisi.perfhub.xml 
/system/lib64/libperfhub_client.so 
/system/lib64/libperfhub.so 
/system/lib64/libperfhub_service.so

The libs above require this one:
adb push  system/lib/libhwuibp.so /system/lib/libhwuibp.so
adb push  system/lib64/libhwuibp.so /system/lib64/libhwuibp.so

There's a missing symbol located in:
adb push  system/lib/libhwui.so /system/lib/libhwui.so
adb push  system/lib64/libhwui.so /system/lib64/libhwui.so

I know from the hwc story that it doesn't cause issues to take the libhwui library but using the vendor library as shim lib is probably the better option (not tested)
```

Nexolight, 2017-01-19:
There's a feature-camera branch now

Nexolight, 2017-01-21:
There's a line in the kernel message:
```
01-21 02:28:32.480 <4>[  770.909851s][2017:01:21 02:28:32][pid:8759,cpu1,ispreq][CAMERA]ERROR: open file error!

```

and there's only a single file in the kernel which has this line:
```
drivers/media/huawei/camera/isp/ovisp2.3/isp_ops.c
```

The function is called load_firmware and takes a pointer to a filename.
So it might be that it just can't find the firmware (?)

Nexolight, 2017-01-21:
Got HwCamCfgSvr to run without crashing immediately and the logs look promising!

AD2076, 2017-01-23:
Have a look in vendor6.0/system/fpgaice40 and import those files.

AD2076, 2017-01-23:
isp.bin is in system/isp
libHW_BShutter system/lib (not in lib64)

Nexolight, 2017-01-23:
thx I will look at it at thursday

Nexolight, 2017-01-30:
I usually look into both lib folders and copy both in case they are there. So the ones you mentioned are already in place Except libHW_BShutter.so which doesn't exist (no issue according to the logs), the files in system/fpgaice40 and the system/isp/isp.bin file. I will try it.

Nexolight, 2017-01-30:
Get a segmentation fault when starting the camera:

```
01-30 18:54:22.750  2929  2929 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-30 18:54:22.750  2929  2929 F DEBUG   : CM Version: '13.0-20170120-UNOFFICIAL-hwgra'
01-30 18:54:22.750  2929  2929 F DEBUG   : Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/3c96ef6e76:userdebug/test-keys'
01-30 18:54:22.750  2929  2929 F DEBUG   : Revision: '0'
01-30 18:54:22.750  2929  2929 F DEBUG   : ABI: 'arm'
01-30 18:54:22.750  2929  2929 F DEBUG   : pid: 4302, tid: 9286, name: Processer  >>> /system/bin/HwCamCfgSvr <<<
01-30 18:54:22.750  3348  3477 W NativeCrashListener: Couldn't find ProcessRecord for pid 4302
01-30 18:54:22.750  2929  2929 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x441abd59
01-30 18:54:22.763  2929  2929 F DEBUG   :     r0 441abd59  r1 441abd59  r2 00000000  r3 00000007
01-30 18:54:22.763  2929  2929 E DEBUG   : AM write failed: Broken pipe
01-30 18:54:22.763  2929  2929 F DEBUG   :     r4 ffffffff  r5 f6e9b698  r6 ebc7a338  r7 ede3e68b
01-30 18:54:22.763  2929  2929 F DEBUG   :     r8 ebc7a334  r9 0000000c  sl eea1e478  fp ebc77d4d
01-30 18:54:22.763  2929  2929 F DEBUG   :     ip 80000000  sp eea1e358  lr f708d9b5  pc f705f916  cpsr a00f0030
01-30 18:54:22.769  2929  2929 F DEBUG   :
01-30 18:54:22.769  2929  2929 F DEBUG   : backtrace:
01-30 18:54:22.769  2929  2929 F DEBUG   :     #00 pc 00018916  /system/lib/libc.so (strlen+21)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #01 pc 000469b1  /system/lib/libc.so (__strlen_chk+4)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #02 pc 0054ee08  /system/vendor/lib/libGLES_mali.so
01-30 18:54:22.769  2929  2929 F DEBUG   :     #03 pc 0054bb9c  /system/vendor/lib/libGLES_mali.so
01-30 18:54:22.769  2929  2929 F DEBUG   :     #04 pc 0054b0a4  /system/vendor/lib/libGLES_mali.so
01-30 18:54:22.769  2929  2929 F DEBUG   :     #05 pc 00545f18  /system/vendor/lib/libGLES_mali.so
01-30 18:54:22.769  2929  2929 F DEBUG   :     #06 pc 00539b14  /system/vendor/lib/libGLES_mali.so
01-30 18:54:22.769  2929  2929 F DEBUG   :     #07 pc 0052af04  /system/vendor/lib/libGLES_mali.so (clCreateProgramWithBinary+276)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #08 pc 000977c0  /system/lib/libmrc_cg_postprocess_cl.so (_Z12BuildProgramPKci+1280)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #09 pc 000979a0  /system/lib/libmrc_cg_postprocess_cl.so (setupOpencl+32)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #10 pc 000b2821  /system/lib/libcamera_algo.so (_ZN7android18DavinciCLLibLoader11getInstanceEPNS_18HwaDev
iceInterfaceE+380)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #11 pc 0007b6fd  /system/lib/libcamera_algo.so (_ZN7android9PPPrepare5startEv+56)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #12 pc 000747b7  /system/lib/libcamera_algo.so (_ZN7android12MsgProcesser10threadLoopEv+32)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #13 pc 00010139  /system/lib/libutils.so (_ZN7android6Thread11_threadLoopEPv+112)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #14 pc 0003fe4b  /system/lib/libc.so (_ZL15__pthread_startPv+30)
01-30 18:54:22.769  2929  2929 F DEBUG   :     #15 pc 00019c85  /system/lib/libc.so (__start_thread+6)
```

Nexolight, 2017-01-30:
Meanwhile in kmsgcat when opening the camera

```
01-30 22:24:36.672 <4>[ 1667.698760s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]====================dump ddr register begin======================
==
01-30 22:24:36.672 <4>[ 1667.698760s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]REG_AXI_INT_STATUS is 0x4
01-30 22:24:36.672 <4>[ 1667.698791s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]REG_QOSB_INTSTS    is 0x0
01-30 22:24:36.672 <4>[ 1667.698791s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]REG_DDRC_INTSTS_DMC0    is 0x0
01-30 22:24:36.672 <4>[ 1667.698822s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]REG_DDRC_INTSTS_DMC1    is 0x0
01-30 22:24:36.672 <4>[ 1667.698822s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]the error port is 0x2
01-30 22:24:36.672 <4>[ 1667.698852s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]REG_SEC_INT_EN    is 0x1
01-30 22:24:36.672 <4>[ 1667.698852s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]REG_SEC_INT_STATUS    is 0x3
01-30 22:24:36.672 <4>[ 1667.698852s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]the fail command0(low 32 address)  is 0x59b00
01-30 22:24:36.672 <4>[ 1667.698883s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]the fail command1(error type)	    is 0x120000
01-30 22:24:36.672 <4>[ 1667.698883s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]the fail command1(error type)      parse:
01-30 22:24:36.672 <4>[ 1667.698913s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]access is unprivileged
01-30 22:24:36.672 <4>[ 1667.698913s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]access is non-secure
01-30 22:24:36.672 <4>[ 1667.698944s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]write operation
01-30 22:24:36.672 <4>[ 1667.698944s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]the fail command2(MasterID) 	    is 0x320000e2
01-30 22:24:36.672 <4>[ 1667.698944s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]the Master is :
01-30 22:24:36.672 <4>[ 1667.698974s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]port is 0x2, mid is 0x32
01-30 22:24:36.672 <4>[ 1667.698974s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]unknow Master
01-30 22:24:36.672 <4>[ 1667.699005s][2017:01:30 22:24:36][pid:3848,cpu0,in irq]====================dump ddr register end========================
==
```

Nexolight, 2017-01-31:
Error when trying to record video
```
01-31 13:57:17.357  3057  3670 E CameraSource: Requested frame rate (15) is not supported: 25,30
01-31 13:57:17.357  3057  3670 I Camera2Client: unlock: E. mClientPid=3057
01-31 13:57:17.357  6529  6529 E MediaRecorder: start failed: -19
01-31 13:57:17.359  2953  6853 I         : [Grace_0.0.5.0][smartAf]af_doAF: flow: 0, giNumFlagDoAF: 51, afType: 1
01-31 13:57:17.359  2953  6858 W CFilterDownscale: virtual hw::TRetCode hw::CFilterDownscale::receive(hw::IBuffer*): instance[0xf474b200], drop.
01-31 13:57:17.359  2953  6853 D AFDeviceImpl: [90af8a5_D]enter af_get_vsync_time()
01-31 13:57:17.360  3057  6845 I Camera3StreamImpl: instance[0xf1d9af80], dequeue buffer(4) succeed, count(2) still at DRV .
01-31 13:57:17.360  2953  6853 D DeviceImpl: [HWA_CAM3]getActualExposure, exp_time=0x752a
01-31 13:57:17.360  2953  6853 D DeviceImpl: [HWA_CAM3]virtual int hw::DeviceImpl::getAEStatus(android::HwaIppFrame*, int*, int*), get from ov-isp, aecStabe=0x0, currentY=0x10e8
01-31 13:57:17.360  2953  6853 I         : [Grace_0.0.5.0][smartAf]AFGetDeviceInfo: ae_gain  987,ae_stable    0, exposure_time 29994 ae_iso  482 fps   29
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: Could not start media recorder.
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: java.lang.RuntimeException: start failed.
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at android.media.MediaRecorder.start(Native Method)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at com.android.camera.VideoModule$11.onStorageUpdateDone(VideoModule.java:1384)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at com.android.camera.CameraActivity$20.onPostExecute(CameraActivity.java:2424)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at com.android.camera.CameraActivity$20.onPostExecute(CameraActivity.java:2419)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at android.os.AsyncTask.finish(AsyncTask.java:651)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at android.os.AsyncTask.-wrap1(AsyncTask.java)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at android.os.AsyncTask$InternalHandler.handleMessage(AsyncTask.java:668)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at android.os.Handler.dispatchMessage(Handler.java:102)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at android.os.Looper.loop(Looper.java:148)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at android.app.ActivityThread.main(ActivityThread.java:5461)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at java.lang.reflect.Method.invoke(Native Method)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
01-31 13:57:17.360  6529  6529 E CAM_VideoModule: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
01-31 13:57:17.360  2953  6853 D AFDeviceImpl: [90af8a5_D]enter af_get_mmi_mode()
01-31 13:57:17.360  2953  6853 D AFDeviceImpl: [90af8a5_D]af_get_mmi_mode() result: 0
01-31 13:57:17.360  2953  6853 I         : [Grace_0.0.5.0][smartAf]AFGetInfo:  xyz:[29,0,-5],iso:482,expotime:29994,gain:987,ae_stable:0,fps:29,iAWBratio:256,iZoom:100,iCameraPosStatus:0,iOperatorSwitch:0
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: Handling Camera Access Failure:
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: java.lang.Exception
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at com.android.camera.FatalErrorHandlerImpl.onGenericCameraAccessFailure(FatalErrorHandlerImpl.java:84)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at com.android.camera.VideoModule$11.onStorageUpdateDone(VideoModule.java:1387)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at com.android.camera.CameraActivity$20.onPostExecute(CameraActivity.java:2424)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at com.android.camera.CameraActivity$20.onPostExecute(CameraActivity.java:2419)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at android.os.AsyncTask.finish(AsyncTask.java:651)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at android.os.AsyncTask.-wrap1(AsyncTask.java)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at android.os.AsyncTask$InternalHandler.handleMessage(AsyncTask.java:668)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at android.os.Handler.dispatchMessage(Handler.java:102)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at android.os.Looper.loop(Looper.java:148)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at android.app.ActivityThread.main(ActivityThread.java:5461)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at java.lang.reflect.Method.invoke(Native Method)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
01-31 13:57:17.361  6529  6529 E CAM_FatalErrorHandler: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
01-31 13:57:17.361  6529  6529 E CAM_CameraUtil: Show fatal error dialog

```

Might be just the wrong framerate request

Nexolight, 2017-01-31:
Using system/etc/media_profiles.xml from stock does help a bit. Now other modes than qvga for video record are selectable and the app doesnt crash + starts "recording". However the screen is frozen at that point.

Nexolight, 2017-01-31:
I close this for now.

closed 2017-01-31 by Nexolight