CM-Huawei-P8-Development/device_gra_l09 #7: Feature: SDCard

Nexolight, 2016-12-16:
SDCard isn't reconized in the system yet. For recovery it is fine.

PaulFasola, 2017-01-02:
Fixed in my Huawei Next's build, I don't remember well but I'll find and PR you the fix ;)

SpeedoWBT, 2017-01-26:
persist.fw.force_adoptable=true
must be  in system.prop (build.prop) to complete the fix as well

Nexolight, 2017-01-27:
It's applied here 42f81fc7245fbbf4a42378657f7ded23819fdc53
I can't test it (don't have a sdcard right now). I will reopen if someone reports issues.

Mosterxxx, 2017-02-01:
Hey Guys,
I have already downloaded last build but i have problem with the sd card, 
At the time of trying to format it Throws this error "Command '19 volume partition disk:179_160 private' failed with '400 19 Command failed', any solution?.

ayesse, 2017-02-02:
Same issue as @Mosterxxx

Nexolight, 2017-02-02:
Does an external format with fat32 fs help?

Well the fix was untested (As I said I don't have an sdcard to test atm)

Mosterxxx, 2017-02-02:
I try to change the sd of exfat to NTFS and ex2 but has failed with any of these.

murdge, 2017-02-03:
For me it works after formatting with SD Card Formatter (exFAT only)
[https://www.sdcard.org/downloads/formatter_4/](url)

Mosterxxx, 2017-02-03:
Thx @murdge , Now works for me, after formatting with you app, but i can't use as primary storage... still needs formatting to use as primary storage.

Nexolight, 2017-02-03:
What about mounting it directly via command line?

I'm not familiar with that stuff. Earlier on other roms I had sdcard and sdcard1 folder.

I need to check the ramdisk can't remember what the fix was but trying to mount it as vfat into /sdcard (I assume that's what you mean with as internal storage) would of curse fail when the sdcard is formated with exFAT.

I'm a bit confused about vold and how it does work. But if it would be just a plain fstab entry I could set the fs to automatic.

Nexolight, 2017-02-03:
The other question is why does it work with exFAT and nothing else ?!?

murdge, 2017-02-04:
There where always problems with P8 and SD Card. You can find more here if you can read german:
[http://www.android-hilfe.de/thema/voller-zugriff-auf-externe-sd-mit-xinternalsd-xposed-modul.783647/](url)
There is also rufus.wilson at xda [https://forum.xda-developers.com/android/software/fix-extsd-fix-v1-0b-2016-01-18-t3296266](url) who tried to make a patch for MM. Maybe it helps to contakt him.

Nexolight, 2017-02-13:
The SDCard itself does work - formating it is what is broken.

closed 2017-02-13 by Nexolight