CM-Huawei-P8-Development/device_gra_l09 #8: MTP does not work

Nexolight, 2016-12-16:
I'm unable to access via MTP so far. Proof me wrong it may just be my frankenstein setup. In recovery mtp works.

Nexolight, 2016-12-27:
I also expirience adb issues sometimes. However executing setprops sys.usb.config adb makes it work again while changeing the option in the developer options don't do anything.

SpeedoWBT, 2017-01-26:
fix MTP : 
http://www.mediafire.com/file/7r00cflrv47n8s8/init.usb.ramdisk_files.zip
and for adb i did this in ramdisk "default.prop"
ro.adb.secure=0
ro.secure=0
security.perf_harden=0
ro.allow.mock.location=0
ro.debuggable=1
ro.zygote=zygote64_32
dalvik.vm.image-dex2oat-Xms=64m
dalvik.vm.image-dex2oat-Xmx=64m
dalvik.vm.dex2oat-Xms=64m
dalvik.vm.dex2oat-Xmx=512m
ro.dalvik.vm.native.bridge=0
debug.atrace.tags.enableflags=0
persist.service.adb.enable=1                                                    
persist.service.debuggable=1
persist.sys.usb.config=mtp,adb

Nexolight, 2017-01-27:
MTP fixed here 45fc168ebe0e7768c30c4a67899d413cd6bfa01a and here 42f81fc7245fbbf4a42378657f7ded23819fdc53

I haven't applied it 1:1
I don't reallly want to replace the gernerated init config. My expirience so far was that this is only possible by providing the full init folder. I rather go along with the build system. I have put your init.usb.rc into a separated init file imported after the cm13 stuff is done.

It does work fine. However USB Tethering doesn't work yet. Maybe I missed a service. I had to name them differently as they don't have the same name for me. The RNDIS is reconized but there are some issues I haven't investigated yet.

closed 2017-01-27 by Nexolight