CM-Huawei-P8-Development/device_gra_l09 #9: Random app crashes

Nexolight, 2016-12-16:
Anyone can help with this one.
There may be something wrong. As example Whatsapp and a lot of other messengers crash frequently. It may be related to a single library or something but to point out what's wrong we need some logs.

Feel free to post the Exceptions or whatever you think may be interesting.

Nexolight, 2016-12-16:
A.I Type

```
12-16 23:21:55.203  7647  7655 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x4 in tid 7655 (FinalizerDaemon)
12-16 23:21:55.203  7369  7369 I DEBUG   : property debug.db.uid not set; NOT waiting for gdb.
12-16 23:21:55.203  7369  7369 I DEBUG   : HINT: adb shell setprop debug.db.uid 100000
12-16 23:21:55.203  7369  7369 I DEBUG   : HINT: adb forward tcp:5039 tcp:5039
12-16 23:21:55.223  7647  7704 W FlurryAgent: Analytics report sent.
12-16 23:21:55.260  2910  4357 I [Gralloc]: alloc w[1080] h[832] format[1] usage[2816]
12-16 23:21:55.262  2910  4357 I [Gralloc]: alloc succ handle[0x7faf250c40] stride[1120]
12-16 23:21:55.305  7369  7369 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
12-16 23:21:55.305  7369  7369 F DEBUG   : CM Version: '13.0-20161216-UNOFFICIAL-hwgra'
12-16 23:21:55.305  7369  7369 F DEBUG   : Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/6ec64bdff6:userdebug/test-keys'
12-16 23:21:55.305  7369  7369 F DEBUG   : Revision: '0'
12-16 23:21:55.305  7369  7369 F DEBUG   : ABI: 'arm'
12-16 23:21:55.305  7369  7369 F DEBUG   : pid: 7647, tid: 7655, name: FinalizerDaemon  >>> com.aitype.android.p <<<
12-16 23:21:55.305  7369  7369 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x4
12-16 23:21:55.330  7369  7369 F DEBUG   :     r0 f4c15400  r1 f51324f8  r2 00000008  r3 00000000
12-16 23:21:55.331  7369  7369 F DEBUG   :     r4 dffc2f00  r5 00000000  r6 dffc2f00  r7 00000000
12-16 23:21:55.331  7369  7369 F DEBUG   :     r8 70f2c2e0  r9 edf5dd00  sl 13062b20  fp 1311cba0
12-16 23:21:55.331  7369  7369 F DEBUG   :     ip f5132048  sp f4502548  lr f70f7caf  pc f6890694  cpsr 800f0030
12-16 23:21:55.332  7369  7369 F DEBUG   :
12-16 23:21:55.332  7369  7369 F DEBUG   : backtrace:
12-16 23:21:55.332  7369  7369 F DEBUG   :     #00 pc 000b4694  /system/lib/libicui18n.so (_ZN6icu_5513DecimalFormatD1Ev+99)
12-16 23:21:55.332  7369  7369 F DEBUG   :     #01 pc 000b470d  /system/lib/libicui18n.so (_ZN6icu_5513DecimalFormatD0Ev+4)
12-16 23:21:55.332  7369  7369 F DEBUG   :     #02 pc 735e8c15  /data/dalvik-cache/arm/system@framework@boot.oat (offset 0x1f75000)
12-16 23:21:55.944  7369  7369 F DEBUG   :
12-16 23:21:55.944  7369  7369 F DEBUG   : Tombstone written to: /data/tombstones/tombstone_08
12-16 23:21:55.944  7369  7369 E DEBUG   : AM write failed: Broken pipe
```

Nexolight, 2016-12-16:
I remember that I've seen a similar stacktrace with whatsapp.

Nexolight, 2017-01-31:
Play store
```
01-31 18:10:33.631  5940  5940 E AndroidRuntime: FATAL EXCEPTION: main
01-31 18:10:33.631  5940  5940 E AndroidRuntime: Process: com.android.vending, PID: 5940
01-31 18:10:33.631  5940  5940 E AndroidRuntime: Theme: themes:{}
01-31 18:10:33.631  5940  5940 E AndroidRuntime: java.lang.OutOfMemoryError: Failed to allocate a 1369452 byte allocation with 840912 free bytes and 821KB until OOM
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.graphics.Bitmap.nativeCreate(Native Method)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.graphics.Bitmap.createBitmap(Bitmap.java:832)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.graphics.Bitmap.createBitmap(Bitmap.java:809)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.graphics.Bitmap.createBitmap(Bitmap.java:776)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.TransitionUtils.createViewBitmap(TransitionUtils.java:165)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.TransitionUtils.copyViewImage(TransitionUtils.java:102)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.Visibility.onDisappear(Visibility.java:384)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.Visibility.createAnimator(Visibility.java:247)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.Transition.createAnimators(Transition.java:732)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.TransitionSet.createAnimators(TransitionSet.java:405)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.TransitionSet.createAnimators(TransitionSet.java:405)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.Transition.playTransition(Transition.java:1777)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.transition.TransitionManager$MultiListener.onPreDraw(TransitionManager.java:306)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.view.ViewTreeObserver.dispatchOnPreDraw(ViewTreeObserver.java:944)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2063)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1115)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6023)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:858)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.view.Choreographer.doCallbacks(Choreographer.java:670)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.view.Choreographer.doFrame(Choreographer.java:606)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:844)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:739)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:95)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:148)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5461)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
01-31 18:10:33.631  5940  5940 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
```

Nexolight, 2017-02-01:
Changing dalvik.vm.heapgrowthlimit=48m to whatever (try with 128m) seems to fix a few issues with oom errors.

Nexolight, 2017-02-04:
Instagram

```
02-03 13:41:40.052 F/libc    (7639): Fatal signal 11 (SIGSEGV), code 1, fault addr 0x4 in tid 7648 (FinalizerDaemon)
02-03 13:41:40.129 F/DEBUG   (2953): *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-03 13:41:40.129 F/DEBUG   (2953): CM Version: '13.0-20170131-UNOFFICIAL-hwgra'
02-03 13:41:40.129 F/DEBUG   (2953): Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/cb30c6d545:userdebug/test-keys'
02-03 13:41:40.129 F/DEBUG   (2953): Revision: '0'
02-03 13:41:40.129 F/DEBUG   (2953): ABI: 'arm'
02-03 13:41:40.130 F/DEBUG   (2953): pid: 7639, tid: 7648, name: FinalizerDaemon  >>> com.instagram.android <<<
02-03 13:41:40.130 F/DEBUG   (2953): signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x4
02-03 13:41:40.168 F/DEBUG   (2953):     r0 d78d0600  r1 e953f4f8  r2 00000008  r3 00000000
02-03 13:41:40.168 F/DEBUG   (2953):     r4 d9d55000  r5 00000000  r6 d9d55000  r7 00000000
02-03 13:41:40.169 F/DEBUG   (2953):     r8 713c6ff0  r9 ec1c9800  sl 331d6100  fp 33233180
02-03 13:41:40.169 F/DEBUG   (2953):     ip e953f048  sp f4744548  lr f74abcaf  pc f6c44694  cpsr 800f0030
02-03 13:41:40.170 F/DEBUG   (2953): 
02-03 13:41:40.170 F/DEBUG   (2953): backtrace:
02-03 13:41:40.170 F/DEBUG   (2953):     #00 pc 000b4694  /system/lib/libicui18n.so (_ZN6icu_5513DecimalFormatD1Ev+99)
02-03 13:41:40.171 F/DEBUG   (2953):     #01 pc 000b470d  /system/lib/libicui18n.so (_ZN6icu_5513DecimalFormatD0Ev+4)
02-03 13:41:40.171 F/DEBUG   (2953):     #02 pc 73a82c15  /data/dalvik-cache/arm/system@framework@boot.oat (offset 0x1f75000)
02-03 13:41:41.047 W/ActivityManager(3334):   Force finishing activity com.instagram.android/.activity.MainTabActivity
02-03 13:41:41.047 F/DEBUG   (2953): 
02-03 13:41:41.047 F/DEBUG   (2953): Tombstone written to: /data/tombstones/tombstone_00
```
From here: https://dl.xda-developers.com/4/0/2/7/6/0/8/2017-02-03-13-41-55.txt?key=YlF4o_tC_gr_6et0rox42Q&ts=1486168122

Exactly the same issue like A.I. Type

Nexolight, 2017-02-06:
Pokemon go... WTF? (that's all)
```
02-06 22:46:37.814 17625 17625 E CRASH   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 00000004
02-06 22:46:37.814 17625 17625 E CRASH   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-06 22:46:37.814 17625 17625 E CRASH   : Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/c10310f96f:userdebug/test-keys'
02-06 22:46:37.814 17625 17625 E CRASH   : Revision: '0'

```

Nexolight, 2017-02-13:
Viper4Android

```
02-13 16:05:10.082  9718  9718 I ViPER4Android: ViPER4Android service connected
02-13 16:05:10.083  9718 10127 D mali_winsys: new_window_surface returns 0x3000
02-13 16:05:10.084  2921  2931 I [Gralloc]: alloc succ handle[0x7fa99d02a0] stride[1248]
02-13 16:05:10.085  2921  2931 I [Gralloc]: alloc w[1194] h[818] format[1] usage[2304]
02-13 16:05:10.087  2921  2931 I [Gralloc]: alloc succ handle[0x7fa99d0460] stride[1248]
02-13 16:05:10.088  2921  2931 I [Gralloc]: alloc w[1194] h[818] format[1] usage[2304]
02-13 16:05:10.091  2921  2931 I [Gralloc]: alloc succ handle[0x7fa99d0a80] stride[1248]
02-13 16:05:10.094  9718  9718 I ViPER4Android: Unbinding service ...
02-13 16:05:10.132  9718  9718 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x4 in tid 9718 (r4android_v2_fx)
02-13 16:05:10.133  2937  2937 I DEBUG   : property debug.db.uid not set; NOT waiting for gdb.
02-13 16:05:10.133  2937  2937 I DEBUG   : HINT: adb shell setprop debug.db.uid 100000
02-13 16:05:10.133  2937  2937 I DEBUG   : HINT: adb forward tcp:5039 tcp:5039
02-13 16:05:10.184  2937  2937 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-13 16:05:10.184  2937  2937 F DEBUG   : CM Version: '13.0-20170206-UNOFFICIAL-hwgra'
02-13 16:05:10.184  2937  2937 F DEBUG   : Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/c10310f96f:userdebug/test-keys'
02-13 16:05:10.184  2937  2937 F DEBUG   : Revision: '0'
02-13 16:05:10.184  2937  2937 F DEBUG   : ABI: 'arm'
02-13 16:05:10.184  2937  2937 F DEBUG   : pid: 9718, tid: 9718, name: r4android_v2_fx  >>> com.vipercn.viper4android_v2_fx <<<
02-13 16:05:10.184  2937  2937 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x4
02-13 16:05:10.200  2937  2937 F DEBUG   :     r0 00000004  r1 ffa3a034  r2 b71ed500  r3 ffffffff
02-13 16:05:10.200  2937  2937 F DEBUG   :     r4 00000000  r5 c4c441b0  r6 c4c46440  r7 c4c34000
02-13 16:05:10.200  2937  2937 F DEBUG   :     r8 c4c46400  r9 f4f36500  sl b71ed500  fp ffffffff
02-13 16:05:10.200  2937  2937 F DEBUG   :     ip f7270c24  sp ffa3a020  lr f726561d  pc f7265602  cpsr 00000030
02-13 16:05:10.201  2937  2937 F DEBUG   :
02-13 16:05:10.201  2937  2937 F DEBUG   : backtrace:
02-13 16:05:10.201  2937  2937 F DEBUG   :     #00 pc 0000e602  /system/lib/libutils.so (_ZN7android7RefBase12weakref_type7incWeakEPKv+5)
02-13 16:05:10.201  2937  2937 F DEBUG   :     #01 pc 0000e619  /system/lib/libutils.so (_ZNK7android7RefBase9incStrongEPKv+6)
02-13 16:05:10.201  2937  2937 F DEBUG   :     #02 pc 0007e361  /system/lib/libandroid_runtime.so (_ZN7android2spINS_7SurfaceEEC2EPS1_+12)
02-13 16:05:10.201  2937  2937 F DEBUG   :     #03 pc 0007e85d  /system/lib/libandroid_runtime.so
02-13 16:05:10.201  2937  2937 F DEBUG   :     #04 pc 71f1bc15  /data/dalvik-cache/arm/system@framework@boot.oat (offset 0x1f75000)
02-13 16:05:10.315  3830  4045 V HextileEngine: RenderTask completed
02-13 16:05:10.504  3344 10149 W ActivityManager:   Force finishing activity com.vipercn.viper4android_v2_fx/.activity.ViPER4Android
02-13 16:05:10.505  2937  2937 F DEBUG   :
02-13 16:05:10.505  2937  2937 F DEBUG   : Tombstone written to: /data/tombstones/tombstone_06
02-13 16:05:10.505  2937  2937 E DEBUG   : AM write failed: Broken pipe
```

Nexolight, 2017-02-13:
Chrome

```
02-13 23:15:28.550  9167  9039 F google-breakpad: M AAD29000 00000000 00005000 2B84F6F82D07C718B24C4B397C733D670 app_process32
02-13 23:15:28.550  9167  9039 F google-breakpad: M 73930000 01F75000 017A7000 000000000000000000000000000000000 system@framework@boot.oat
02-13 23:15:28.550  9167  9039 F google-breakpad: M 9EAEC000 009A8000 029F6000 0E2314DF59EAF00F5D5E380F955EDBE90 libchrome.so
02-13 23:15:28.550  9167  9039 F google-breakpad: M B5249000 00000000 00004000 1FB9E55F4CC476177053644E3B0D76D00 libwebviewchromium_loader.so
02-13 23:15:28.550  9167  9039 F google-breakpad: M B52ED000 00000000 00003000 1ACCBEF087A03D534CE67B40F6FE9D4D0 libjnigraphics.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B52F2000 00000000 00008000 3265960CB814B66D4EF5E56143D516950 libcompiler_rt.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B52FB000 00000000 0000F000 5C167EF4DB2D9A6226CB6FFF040E8FF60 libandroid.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B526D000 00000000 01668000 EF7869D3D48A8F1EA901D86C23AA7DB30 libGLES_mali.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7EF5000 00000000 00018000 E260851B5AD44E1BF37DA82E6D676C5E0 libkeymaster1.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7F10000 00000000 00009000 FD0691033E6423739937A99376499D560 libkeymaster_messages.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7F1B000 00000000 0000F000 B2D2FEE4EC94890D1BAD5056962CE97C0 libsoftkeymasterdevice.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7F2C000 00000000 0000B000 A94A2307919D7C2843921A7072C70D820 libkeystore_binder.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7F3A000 00000000 00004000 17545A9B117D9A2386B7A31EDBA286390 libkeystore-engine.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7F81000 00000000 00017000 82217C562A3BAA0072C52509B3D85AE40 libjavacrypto.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7F9A000 00000000 00004000 15EDF94219C16C04CCD2C23FD3BB62820 libemoji.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7FA0000 00000000 0001B000 D171897F073574B9B7C8ED0DF47BE6880 libswresample.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7FBE000 00000000 00008000 30A799D526B9C8B3A8EBD793D192555E0 libffmpeg_utils.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B7FC7000 00000000 00059000 51AC5FAD072F2CCE79766EFE63ED677D0 libavutil.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B8038000 00000000 001B1000 5C6D2BB1F46A053530BD26A4EA57ACB80 libavformat.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B81EC000 00000000 00AB0000 343ADB338885BF4A094DA09615CAB21C0 libavcodec.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B9399000 00000000 0000D000 2181865B3483D2CB95467370E45F150E0 libffmpeg_extractor.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B93A8000 00000000 0000E000 7F0C02731076CAA00C884C9AE74284100 libstagefright_amrnb_common.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B93B5000 00000000 00022000 0C23916F595675EFE51209332DEE1F1F0 libexif.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B93E4000 00000000 00015000 98CAE877C2FC69E51977D53E921EAF220 libmtp.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M B93FA000 00000000 00046000 D1586441291EE2A82E8709DA307036A00 libmedia_jni.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M BCE0B000 00000000 00003000 CB976F871F44CAE247F64440172389380 memtrack.hi3635.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M BE87B000 00000000 00034000 E318E68E83ECA9EF9341204B95308EE40 libjavacore.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M E25CF000 00000000 00004000 25D7B598FB5B166F4FC80D9F0466562D0 libion.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M E25D5000 00000000 00008000 3E6BB618915D9AAE7A037CD6A8B31D080 gralloc.hi3635.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M EC990000 00000000 00013000 EB1023E2F4B3DE0E85F5221BA04D24CA0 libchromium_android_linker.so
02-13 23:15:28.551  9167  9039 F google-breakpad: M F058F000 03208000 0314D000 000000000000000000000000000000000 base.odex
02-13 23:15:28.551  9167  9039 F google-breakpad: M F36E2000 00004000 00002000 000000000000000000000000000000000 system@framework@com.android.media.remotedisplay.jar@classes.dex
02-13 23:15:28.552  9167  9039 F google-breakpad: M F36EF000 0000A000 00005000 000000000000000000000000000000000 system@framework@com.android.location.provider.jar@classes.dex
02-13 23:15:28.552  9167  9039 F google-breakpad: M F424B000 007E8000 006F3000 000000000000000000000000000000000 base.odex
02-13 23:15:28.552  9167  9039 F google-breakpad: M F49F2000 00000000 00441000 DF8C3A3FBAE4CB17C9EA090BA90444E10 libart.so
02-13 23:15:28.552  9167  9039 F google-breakpad: M F4F6B000 00000000 0006F000 0B3745BF33144F77057B6208426217310 libshim_hwui.so
02-13 23:15:28.552  9167  9039 F google-breakpad: M F4FC6000 00000000 000AD000 AFFA608F26C4DD977F55555997AB04B50 libshim_media.so
02-13 23:15:28.552  9167  9039 F google-breakpad: M F508B000 00000000 0017A000 D5963264DD450B09D85C7A2328A646650 libshim_icui18n.so
02-13 23:15:28.552  9167  9039 F google-breakpad: M F520B000 00000000 0002F000 59D4504A4A908DF7DEC7B638343118D00 libshim_camera_client.so
02-13 23:15:28.552  9167  9039 F google-breakpad: M F523C000 00000000 00056000 6A8A89BFDF9E1755A510FA1EE4F77C470 libshim_gui.so
02-13 23:15:28.552  9167  9039 F google-breakpad: M F527C000 00000000 00A35000 0867A007F9C0EBCE84C32EB48CB644780 libLLVM.so
[... cut ...]
02-13 23:15:28.649  2940  2940 F DEBUG   :     #03 pc 0007e7a3  /system/lib/libandroid_runtime.so
02-13 23:15:28.649  2940  2940 F DEBUG   :     #04 pc 73bd0fbd  /data/dalvik-cache/arm/system@framework@boot.oat (offset 0x1f75000)
02-13 23:15:28.929  3085  3085 D gpsd    : WakeLock(Acquire,GPSD)
02-13 23:15:28.929  3085  3085 D gpsd    : WakeLock(Release,GPSD)
02-13 23:15:29.230  2940  2940 F DEBUG   :
02-13 23:15:29.230  2940  2940 F DEBUG   : Tombstone written to: /data/tombstones/tombstone_04

```

Nexolight, 2017-02-13:
More human readable Viper4Android & Chrome
```
#00 android::RefBase::weakref_type::incWeak(void const*)
#01 android::RefBase::incStrong(void const*) const 
#02 android::sp<android::Surface>::sp(android::Surface*)
```

Nexolight, 2017-02-15:
Facebook
```
F/DEBUG   ( 2927): *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
F/DEBUG   ( 2927): CM Version: '13.0-20170206-UNOFFICIAL-hwgra'
F/DEBUG   ( 2927): Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/c10310f96f:userdebug/test-keys'
F/DEBUG   ( 2927): Revision: '0'
F/DEBUG   ( 2927): ABI: 'arm'
F/DEBUG   ( 2927): pid: 32577, tid: 32577, name: facebook.katana  >>> com.facebook.katana <<<
F/DEBUG   ( 2927): signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xfffffff4
I/debuggerd( 2927): type=1400 audit(0.0:1314): avc: denied { read } for ppid=1 ppid_comm="init" name="libyoga.so" dev="mmcblk0p40" ino=431171 scontext=u:r:debuggerd:s0 tcontext=u:object_r:app_data_file:s0:c512,c768 tclass=file permissive=1
I/debuggerd( 2927): type=1400 audit(0.0:1315): avc: denied { open } for ppid=1 ppid_comm="init" path="/data/data/com.facebook.katana/lib-xzs/libyoga.so" dev="mmcblk0p40" ino=431171 scontext=u:r:debuggerd:s0 tcontext=u:object_r:app_data_file:s0:c512,c768 tclass=file permissive=1
I/debuggerd( 2927): type=1400 audit(0.0:1316): avc: denied { getattr } for ppid=1 ppid_comm="init" path="/data/data/com.facebook.katana/lib-xzs/libyoga.so" dev="mmcblk0p40" ino=431171 scontext=u:r:debuggerd:s0 tcontext=u:object_r:app_data_file:s0:c512,c768 tclass=file permissive=1
F/DEBUG   ( 2927):     r0 94ac8200  r1 94ac8200  r2 00000000  r3 00000001
F/DEBUG   ( 2927):     r4 ff89c664  r5 0000025c  r6 00000438  r7 ff89c68c
F/DEBUG   ( 2927):     r8 0000025c  r9 f4cf6500  sl 0000043e  fp 0000025f
F/DEBUG   ( 2927):     ip 00430000  sp ff89c658  lr f4a2e313  pc f6f66a5e  cpsr 600e0030
F/DEBUG   ( 2927): 
F/DEBUG   ( 2927): backtrace:
F/DEBUG   ( 2927):     #00 pc 00098a5e  /system/lib/libandroid_runtime.so (_ZN7android32SurfaceTexture_getSurfaceTextureEP7_JNIEnvP8_jobject+25)
F/DEBUG   ( 2927):     #01 pc 00098f6f  /system/lib/libandroid_runtime.so
F/DEBUG   ( 2927):     #02 pc 7397424f  /data/dalvik-cache/arm/system@framework@boot.oat (offset 0x1f75000)
```

murdge, 2017-02-15:
#1. Whatsapp try open attached video
#2. CM Camera, trying a 2nd picture
#3. Open Camera, trying a 2nd picture
Last # is given write rights for SD Card to MatLog
```
--------- beginning of system
--------- beginning of crash
02-14 17:07:00.572 F/libc    (12080): Fatal signal 11 (SIGSEGV), code 1, fault addr 0x4 in tid 12088 (FinalizerDaemon)
02-14 17:07:00.625 F/DEBUG   (2933): *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-14 17:07:00.625 F/DEBUG   (2933): CM Version: '13.0-20170206-UNOFFICIAL-hwgra'
02-14 17:07:00.625 F/DEBUG   (2933): Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/c10310f96f:userdebug/test-keys'
02-14 17:07:00.625 F/DEBUG   (2933): Revision: '0'
02-14 17:07:00.625 F/DEBUG   (2933): ABI: 'arm'
02-14 17:07:00.625 F/DEBUG   (2933): pid: 12080, tid: 12088, name: FinalizerDaemon  >>> com.whatsapp <<<
02-14 17:07:00.625 F/DEBUG   (2933): signal 6 (SIGABRT), code 1 (?), fault addr --------
02-14 17:07:00.680 F/DEBUG   (2933):     r0 00000000  r1 00002f38  r2 00000006  r3 c44c2978
02-14 17:07:00.680 F/DEBUG   (2933):     r4 c44c2980  r5 c44c2930  r6 0000000b  r7 0000010c
02-14 17:07:00.680 F/DEBUG   (2933):     r8 c43bdc4c  r9 f507f840  sl c43bdc48  fp c43bdc34
02-14 17:07:00.680 F/DEBUG   (2933):     ip 00000006  sp c43bdba0  lr f725554d  pc f725793c  cpsr 400f0010
02-14 17:07:00.693 F/DEBUG   (2933): 
02-14 17:07:00.693 F/DEBUG   (2933): backtrace:
02-14 17:07:00.693 F/DEBUG   (2933):     #00 pc 0004293c  /system/lib/libc.so (tgkill+12)
02-14 17:07:00.693 F/DEBUG   (2933):     #01 pc 00040549  /system/lib/libc.so (pthread_kill+32)
02-14 17:07:00.693 F/DEBUG   (2933):     #02 pc 0001c44f  /system/lib/libc.so (raise+10)
02-14 17:07:00.693 F/DEBUG   (2933):     #03 pc 00019601  /system/lib/libc.so (__libc_android_abort+34)
02-14 17:07:00.693 F/DEBUG   (2933):     #04 pc 00017514  /system/lib/libc.so (abort+4)
02-14 17:07:00.693 F/DEBUG   (2933):     #05 pc 001f0809  /data/app/com.whatsapp-1/lib/arm/libwhatsapp.so
02-14 17:07:00.693 F/DEBUG   (2933):     #06 pc 003565cb  /system/lib/libart.so (_ZN3art12FaultManager11HandleFaultEiP7siginfoPv+510)
02-14 17:07:00.693 F/DEBUG   (2933):     #07 pc 00017578  /system/lib/libc.so
02-14 17:07:01.832 F/DEBUG   (2933): 
02-14 17:07:01.832 F/DEBUG   (2933): Tombstone written to: /data/tombstones/tombstone_00
02-14 17:21:35.758 F/libc    (3039): Fatal signal 11 (SIGSEGV), code 1, fault addr 0x1723d9bf in tid 3039 (mediaserver)
02-14 17:21:35.810 F/DEBUG   (2933): *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-14 17:21:35.810 F/DEBUG   (2933): CM Version: '13.0-20170206-UNOFFICIAL-hwgra'
02-14 17:21:35.810 F/DEBUG   (2933): Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/c10310f96f:userdebug/test-keys'
02-14 17:21:35.810 F/DEBUG   (2933): Revision: '0'
02-14 17:21:35.810 F/DEBUG   (2933): ABI: 'arm'
02-14 17:21:35.810 F/DEBUG   (2933): pid: 3039, tid: 3039, name: mediaserver  >>> /system/bin/mediaserver <<<
02-14 17:21:35.810 F/DEBUG   (2933): signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x1723d9bf
02-14 17:21:35.821 F/DEBUG   (2933):     r0 ffffff7c  r1 ffffffff  r2 f7250384  r3 1fffffff
02-14 17:21:35.821 F/DEBUG   (2933):     r4 00000000  r5 f199cff0  r6 f6682000  r7 00010000
02-14 17:21:35.821 F/DEBUG   (2933):     r8 f6689008  r9 f7243590  sl 00000001  fp ff9f7a60
02-14 17:21:35.821 F/DEBUG   (2933):     ip f723d9c0  sp ff9f7908  lr f7225caf  pc f7222e8c  cpsr 20000030
02-14 17:21:35.851 F/DEBUG   (2933): 
02-14 17:21:35.851 F/DEBUG   (2933): backtrace:
02-14 17:21:35.851 F/DEBUG   (2933):     #00 pc 00055e8c  /system/lib/libc.so (ifree+399)
02-14 17:21:35.851 F/DEBUG   (2933):     #01 pc 00058cab  /system/lib/libc.so (je_free+374)
02-14 17:21:35.851 F/DEBUG   (2933):     #02 pc 0000e91f  /system/lib/libutils.so (_ZNK7android12SharedBuffer7releaseEj+36)
02-14 17:21:35.851 F/DEBUG   (2933):     #03 pc 0000ed75  /system/lib/libutils.so (_ZN7android7String8D1Ev+18)
02-14 17:21:35.851 F/DEBUG   (2933):     #04 pc 000058e1  /system/vendor/lib/libwvm.so (_ZN7android15IsWidevineMediaERKNS_2spINS_10DataSourceEEE+136)
02-14 17:21:35.851 F/DEBUG   (2933):     #05 pc 000d88a5  /system/lib/libstagefright.so (_ZN7android8SniffWVMERKNS_2spINS_10DataSourceEEEPNS_7String8EPfPNS0_INS_8AMessageEEE+64)
02-14 17:21:35.851 F/DEBUG   (2933):     #06 pc 00091a6d  /system/lib/libstagefright.so (_ZN7android10DataSource5sniffEPNS_7String8EPfPNS_2spINS_8AMessageEEE+216)
02-14 17:21:35.851 F/DEBUG   (2933):     #07 pc 000b826b  /system/lib/libstagefright.so (_ZN7android14MediaExtractor6CreateERKNS_2spINS_10DataSourceEEEPKcjPKNS1_INS_8AMessageEEE+58)
02-14 17:21:35.851 F/DEBUG   (2933):     #08 pc 000ce5e3  /system/lib/libstagefright.so (_ZN7android28StagefrightMetadataRetriever13setDataSourceEixx+126)
02-14 17:21:35.851 F/DEBUG   (2933):     #09 pc 000493eb  /system/lib/libmediaplayerservice.so (_ZN7android23MetadataRetrieverClient13setDataSourceEixx+230)
02-14 17:21:35.851 F/DEBUG   (2933):     #10 pc 0008587f  /system/lib/libmedia.so (_ZN7android24BnMediaMetadataRetriever10onTransactEjRKNS_6ParcelEPS1_j+394)
02-14 17:21:35.852 F/DEBUG   (2933):     #11 pc 000199c9  /system/lib/libbinder.so (_ZN7android7BBinder8transactEjRKNS_6ParcelEPS1_j+60)
02-14 17:21:35.852 F/DEBUG   (2933):     #12 pc 0001ed53  /system/lib/libbinder.so (_ZN7android14IPCThreadState14executeCommandEi+550)
[... cut ...]
02-15 13:37:54.871 F/DEBUG   (2934):     #00 pc 0000000000069ed4  /system/lib64/libc.so (write+8)
02-15 13:37:54.871 F/DEBUG   (2934):     #01 pc 00000000000097f8  /system/xbin/su
02-15 13:37:54.871 F/DEBUG   (2934):     #02 pc 0000000000008598  /system/xbin/su
02-15 13:37:54.871 F/DEBUG   (2934):     #03 pc 000000000001bcd8  /system/lib64/libc.so (__libc_init+100)
02-15 13:37:54.871 F/DEBUG   (2934):     #04 pc 00000000000079fc  /system/xbin/su
02-15 13:37:54.938 F/DEBUG   (2934): 
02-15 13:37:54.938 F/DEBUG   (2934): Tombstone written to: /data/tombstones/tombstone_07

```

murdge, 2017-02-15:
@Nexolight please, can you post the RefBase.cpp?
There was a rare issue with decWeak and incWeak in some RefBase.cpp
```
------thread 1 -----------------
    while (curCount > 0 && curCount != INITIAL_STRONG_VALUE) {
        if (android_atomic_cmpxchg(curCount, curCount+1, &impl->mStrong) == 0) {
            break;
        }
        curCount = impl->mStrong;
    }

-------thread 2----------

refbase->incStrong();  ------->now the strong is 1
refbase->decStrong(); -------> then decrease to 0, and a delete refbase occured
--------thread 1---------
    if (curCount <= 0 || curCount == INITIAL_STRONG_VALUE) {
```

that is the case.

Nexolight, 2017-02-15:
https://github.com/CyanogenMod/android_system_core/blob/stable/cm-13.0-ZNH5Y/libutils/RefBase.cpp

murdge, 2017-02-16:
This RefBase is changed.

The old was
```
    if (curCount <= 0 || curCount == INITIAL_STRONG_VALUE) {
		
        bool allow;
				
        if (curCount == INITIAL_STRONG_VALUE) {
            // Attempting to acquire first strong reference...  this is allowed
            // if the object does NOT have a longer lifetime (meaning the
            // implementation doesn't need to see this), or if the implementation
            // allows it to happen.
            allow = (impl->mFlags&OBJECT_LIFETIME_WEAK) != OBJECT_LIFETIME_WEAK
                  || impl->mBase->onIncStrongAttempted(FIRST_INC_STRONG, id);
        } else {
            // Attempting to revive the object...  this is allowed
            // if the object DOES have a longer lifetime (so we can safely
            // call the object with only a weak ref) and the implementation
            // allows it to happen.
            allow = (impl->mFlags&OBJECT_LIFETIME_WEAK) == OBJECT_LIFETIME_WEAK
                  && impl->mBase->onIncStrongAttempted(FIRST_INC_STRONG, id);
        }
        if (!allow) {
            decWeak(id);
            return false;
        }
        curCount = android_atomic_inc(&impl->mStrong);

        // If the strong reference count has already been incremented by
        // someone else, the implementor of onIncStrongAttempted() is holding
        // an unneeded reference.  So call onLastStrongRef() here to remove it.
        // (No, this is not pretty.)  Note that we MUST NOT do this if we
        // are in fact acquiring the first reference.
        if (curCount > 0 && curCount < INITIAL_STRONG_VALUE) {
            impl->mBase->onLastStrongRef(id);
        }
    }
```
that's the new
```
    if (curCount <= 0 || curCount == INITIAL_STRONG_VALUE) {
        // we're now in the harder case of either:
        // - there never was a strong reference on us
        // - or, all strong references have been released
        if ((impl->mFlags&OBJECT_LIFETIME_WEAK) == OBJECT_LIFETIME_STRONG) {
            // this object has a "normal" life-time, i.e.: it gets destroyed
            // when the last strong reference goes away
            if (curCount <= 0) {
                // the last strong-reference got released, the object cannot
                // be revived.
                decWeak(id);
                return false;
            }

            // here, curCount == INITIAL_STRONG_VALUE, which means
            // there never was a strong-reference, so we can try to
            // promote this object; we need to do that atomically.
            while (curCount > 0) {
                if (android_atomic_cmpxchg(curCount, curCount + 1,
                        &impl->mStrong) == 0) {
                    break;
                }
                // the strong count has changed on us, we need to re-assert our
                // situation (e.g.: another thread has inc/decStrong'ed us)
                curCount = impl->mStrong;
            }

            if (curCount <= 0) {
                // promote() failed, some other thread destroyed us in the
                // meantime (i.e.: strong count reached zero).
                decWeak(id);
                return false;
            }
        } else {
            // this object has an "extended" life-time, i.e.: it can be
            // revived from a weak-reference only.
            // Ask the object's implementation if it agrees to be revived
            if (!impl->mBase->onIncStrongAttempted(FIRST_INC_STRONG, id)) {
                // it didn't so give-up.
                decWeak(id);
                return false;
            }
            // grab a strong-reference, which is always safe due to the
            // extended life-time.
            curCount = android_atomic_inc(&impl->mStrong);
        }

        // If the strong reference count has already been incremented by
        // someone else, the implementor of onIncStrongAttempted() is holding
        // an unneeded reference.  So call onLastStrongRef() here to remove it.
        // (No, this is not pretty.)  Note that we MUST NOT do this if we
        // are in fact acquiring the first reference.
        if (curCount > 0 && curCount < INITIAL_STRONG_VALUE) {
            impl->mBase->onLastStrongRef(id);
        }
    }
```

murdge, 2017-02-16:
I think the last #if (code line 583) we must change to:

```
if (curCount != 0 && curCount != INITIAL_STRONG_VALUE) {
    impl->mBase->onLastStrongRef(id);
 }
```
I can't test it but I realy think that's it. Maybe someone can do this and give us a replay.
They have changed it the same way on CM14.1 and they think it was wrong at CM13 too.
[https://github.com/CyanogenMod/android_system_core/commit/1b07c95b74d9d7f339317f1ed3ed09f1018ab77e](url)

I have compared CM13 RefBase.cpp and CM14.1 RefBase.cpp
I think we can use the CM14.1 RefBase.cpp too without changing anything.
Anyway, to change line 583 in CM13 RefBase.cpp will do it to us.

Nexolight, 2017-02-16:
Thx I will try it next time when I work on it in case no one else tries it before.

Nexolight, 2017-02-21:
Ok I changed what they did in the commit. So far I couldn't provocate any chrome crash anymore but those do still crash and the 2nd one with the depending stacktrace.

Threema
```
02-21 19:41:02.392  2926  3196 I [Gralloc]: alloc succ handle[0x7f7e251420] stride[1120]
02-21 19:41:02.397  8070  8895 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x28 in tid 8895 (RenderThread)
02-21 19:41:02.397  2939  2939 I DEBUG   : property debug.db.uid not set; NOT waiting for gdb.
02-21 19:41:02.397  2939  2939 I DEBUG   : HINT: adb shell setprop debug.db.uid 100000
02-21 19:41:02.397  2939  2939 I DEBUG   : HINT: adb forward tcp:5039 tcp:5039
02-21 19:41:02.448  2939  2939 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-21 19:41:02.448  2939  2939 F DEBUG   : CM Version: '13.0-20170206-UNOFFICIAL-hwgra'
02-21 19:41:02.448  2939  2939 F DEBUG   : Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/c10310f96f:userdebug/test-keys'
02-21 19:41:02.448  2939  2939 F DEBUG   : Revision: '0'
02-21 19:41:02.448  2939  2939 F DEBUG   : ABI: 'arm'
02-21 19:41:02.449  2939  2939 F DEBUG   : pid: 8070, tid: 8895, name: RenderThread  >>> ch.threema.app <<<
02-21 19:41:02.449  2939  2939 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x28
02-21 19:41:02.467  2939  2939 F DEBUG   :     r0 b7273500  r1 00000009  r2 acb5c79c  r3 00000000
02-21 19:41:02.468  2939  2939 F DEBUG   :     r4 bd182000  r5 acb5c7d0  r6 f70040e1  r7 f72c7f78
02-21 19:41:02.468  2939  2939 F DEBUG   :     r8 acb5c7f0  r9 acb5c7f0  sl 448b93c4  fp bd185580
02-21 19:41:02.468  2939  2939 F DEBUG   :     ip f72cc518  sp acb5c798  lr f61dcdc1  pc f70040e8  cpsr a00f0030
02-21 19:41:02.470  2939  2939 F DEBUG   :
02-21 19:41:02.470  2939  2939 F DEBUG   : backtrace:
02-21 19:41:02.471  2939  2939 F DEBUG   :     #00 pc 000490e8  /system/lib/libgui.so (_ZN7android7Surface10hook_queryEPK13ANativeWindowiPi+7)
02-21 19:41:02.471  2939  2939 F DEBUG   :     #01 pc 0001adbf  /system/lib/libhwui.so
02-21 19:41:02.471  2939  2939 F DEBUG   :     #02 pc 0001af31  /system/lib/libhwui.so
02-21 19:41:02.471  2939  2939 F DEBUG   :     #03 pc 0001e791  /system/lib/libhwui.so (_ZN7android10uirenderer12renderthread12RenderThread22dispatchFrameCallbacksEv+164)
02-21 19:41:02.471  2939  2939 F DEBUG   :     #04 pc 0001e857  /system/lib/libhwui.so (_ZN7android10uirenderer12renderthread12RenderThread10threadLoopEv+62)
02-21 19:41:02.471  2939  2939 F DEBUG   :     #05 pc 00010131  /system/lib/libutils.so (_ZN7android6Thread11_threadLoopEPv+112)
02-21 19:41:02.471  2939  2939 F DEBUG   :     #06 pc 000606a3  /system/lib/libandroid_runtime.so (_ZN7android14AndroidRuntime15javaThreadShellEPv+70)
02-21 19:41:02.471  2939  2939 F DEBUG   :     #07 pc 0003fe4b  /system/lib/libc.so (_ZL15__pthread_startPv+30)
02-21 19:41:02.471  2939  2939 F DEBUG   :     #08 pc 00019c85  /system/lib/libc.so (__start_thread+6)
02-21 19:41:02.921  3412  8931 W ActivityManager:   Force finishing activity ch.threema.app/.activities.ComposeMessageActivity
02-21 19:41:02.921  2939  2939 F DEBUG   :
02-21 19:41:02.921  2939  2939 F DEBUG   : Tombstone written to: /data/tombstones/tombstone_08
02-21 19:41:02.921  2939  2939 E DEBUG   : AM write failed: Broken pipe
02-21 19:41:02.922  3412  3489 I BootReceiver: Copying /data/tombstones/tombstone_08 to DropBox (SYSTEM_TOMBSTONE)
02-21 19:41:02.942  3412  8931 W ActivityManager:   Force finishing activity ch.threema.app/.activities.HomeActivity
02-21 19:41:02.962  2924  2924 E lowmemorykiller: Error writing /proc/8070/oom_score_adj; errno=22
```

Viper4Android:
```
2-21 19:43:10.671  2939  2939 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-21 19:43:10.671  2939  2939 F DEBUG   : CM Version: '13.0-20170206-UNOFFICIAL-hwgra'
02-21 19:43:10.671  2939  2939 F DEBUG   : Build fingerprint: 'HUAWEI/cm_hwgra/hwgra:6.0.1/MOB31K/c10310f96f:userdebug/test-keys'
02-21 19:43:10.671  2939  2939 F DEBUG   : Revision: '0'
02-21 19:43:10.671  2939  2939 F DEBUG   : ABI: 'arm'
02-21 19:43:10.671  2939  2939 F DEBUG   : pid: 8320, tid: 8320, name: r4android_v2_fx  >>> com.vipercn.viper4android_v2_fx <<<
02-21 19:43:10.672  2939  2939 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x4
02-21 19:43:10.687  2939  2939 F DEBUG   :     r0 00000004  r1 fff936c4  r2 b7273500  r3 ffffffff
02-21 19:43:10.687  2939  2939 F DEBUG   :     r4 00000000  r5 c4cc2340  r6 c4cc7440  r7 c4cb5000
02-21 19:43:10.687  2939  2939 F DEBUG   :     r8 c4cc7400  r9 f4ff6500  sl b7273500  fp ffffffff
02-21 19:43:10.687  2939  2939 F DEBUG   :     ip f732ac24  sp fff936b0  lr f731f61d  pc f731f602  cpsr 00000030
02-21 19:43:10.688  2939  2939 F DEBUG   :
02-21 19:43:10.688  2939  2939 F DEBUG   : backtrace:
02-21 19:43:10.688  2939  2939 F DEBUG   :     #00 pc 0000e602  /system/lib/libutils.so (_ZN7android7RefBase12weakref_type7incWeakEPKv+5)
02-21 19:43:10.688  2939  2939 F DEBUG   :     #01 pc 0000e619  /system/lib/libutils.so (_ZNK7android7RefBase9incStrongEPKv+6)
02-21 19:43:10.688  2939  2939 F DEBUG   :     #02 pc 0007e361  /system/lib/libandroid_runtime.so (_ZN7android2spINS_7SurfaceEEC2EPS1_+12)
02-21 19:43:10.688  2939  2939 F DEBUG   :     #03 pc 0007e85d  /system/lib/libandroid_runtime.so
02-21 19:43:10.689  2939  2939 F DEBUG   :     #04 pc 72712c15  /data/dalvik-cache/arm/system@framework@boot.oat (offset 0x1f75000)
02-21 19:43:10.740  5823  6246 V HextileEngine: RenderTask completed
02-21 19:43:10.975  3412  9133 W ActivityManager:   Force finishing activity com.vipercn.viper4android_v2_fx/.activity.ViPER4Android
02-21 19:43:10.975  2939  2939 F DEBUG   :
02-21 19:43:10.975  2939  2939 F DEBUG   : Tombstone written to: /data/tombstones/tombstone_07
02-21 19:43:10.975  2939  2939 E DEBUG   : AM write failed: Broken pipe
```

SpeedoWBT, 2017-03-02:
Hi , here is the link to Meticulus Github.
https://github.com/Meticulus/android_device_huawei_hi6250
the files i used are from "cm-13.0-**20170111**-METICULUS-hi6250" rom
download is here : 
http://www.meticulus.co.vu/p/cyanogenmod-for-huawei-p9-lite.html

Nexolight, 2017-03-27:
7ceecb8585afa1896b175da2be5b1d54dd13496e fixed any appcrashes related to the comments here.

closed 2017-03-27 by Nexolight