CM-Huawei-P8-Development/device_gra_l09 #12: Recording video is broken.

Nexolight, 2017-01-31:
At the moment it's impossible to record videos with the camera or screenrecorder apps. Some hint's may be left in issue #2 

I suspect codec issues.

murdge, 2017-02-03:
There are some Mobiles CM13 record videos is still broken like Samsung I9300 S3 (I have it) and some Motorola.
Maybe the same with P8.
Here you find more [https://review.cyanogenmod.org/#/q/Iee07ed1147713a6397fff43e3bbd4f4ed08b56b7](url)

Nexolight, 2017-02-03:
The property which they mention is interesting
It should be already merged but I remember that video recording wasn't 
broken at the very beginning (Screenrecorder)

murdge, 2017-02-06:
Do you know this wrapper for the camera?
[https://github.com/debbiche/android_device_huawei_gra_l09/blob/master/camera/CameraWrapper.cpp](url)

Nexolight, 2017-02-07:
I had it previously in my local build without knowing exactly what it does. But it worked without it and I'm pretty sure the video recording issue is not a camera issue. I even had some broken mp4 files when I messed a bit with the vendor codec files. Normal screenrecording has issues as well.

Nexolight, 2017-02-13:
I reverted the HWC fix and that helps for screencast and various app crashes but not for video recording with camera. It's basically still the same issue.

But this could be a hint. It happens after I try to open the cam for recording the 2nd time. The first time the log is flooded so I can't see what's wrong.
```
02-13 18:59:58.587  3056  3056 E FFmpegExtractor: android-source:0xf1b6a780: avformat_open_input failed, err:Invalid data found when processing input
02-13 18:59:58.588  3056  3056 W FFmpegExtractor: sniff through BetterSniffFFMPEG failed, try LegacySniffFFMPEG
02-13 18:59:58.589  3056  3056 E FFmpegExtractor: android-source:0xf1b6a780|file:/data/media/0/DCIM/Camera/VID_20170131_165835.mp4: avformat_open_input failed, err:Invalid data fo
und when processing input
02-13 18:59:58.589  3056  6244 I Camera3StreamImpl: instance[0xf1338480], enqueue buffer(0) succeed, count(1) still at DRV, timestamp(859435 us).
02-13 18:59:58.589  3056  3056 D FFmpegExtractor: SniffFFMPEG failed to sniff this source
02-13 18:59:58.589  3056  3458 I FFmpegExtractor: android-source:0xe9de03b0
02-13 18:59:58.590  3056  6244 I Camera3MetadataImpl: instance[0xf600a0c0], enqueue metadata(0) succeed, count(1) still at DRV, timestamp(859435 us).
02-13 18:59:58.590  3056  3458 I FFMPEG  : Last message repeated 1 times
02-13 18:59:58.590  3056  3458 I FFMPEG  : [mov,mp4,m4a,3gp,3g2,mj2 @ 0xe986f600] moov atom not found
02-13 18:59:58.590  3056  3458 E FFmpegExtractor: android-source:0xe9de03b0: avformat_open_input failed, err:Invalid data found when processing input
02-13 18:59:58.590  3056  3458 W FFmpegExtractor: sniff through BetterSniffFFMPEG failed, try LegacySniffFFMPEG
02-13 18:59:58.590  3056  3708 E MetadataRetrieverClient: failed to capture a video frame
02-13 18:59:58.591  6211  6274 E MediaMetadataRetrieverJNI: getFrameAtTime: videoFrame is a NULL pointer
02-13 18:59:58.591  3056  3458 E FFmpegExtractor: android-source:0xe9de03b0|file:/data/media/0/DCIM/Camera/VID_20170131_165835.mp4: avformat_open_input failed, err:Invalid data fo
und when processing input
02-13 18:59:58.591  3056  3458 D FFmpegExtractor: SniffFFMPEG failed to sniff this source
02-13 18:59:58.591  3056  6244 I Camera3StreamImpl: instance[0xf1338480], enqueue buffer(1) succeed, count(2) still at DRV, timestamp(881489 us).
02-13 18:59:58.592  3056  3396 E MetadataRetrieverClient: failed to capture a video frame
02-13 18:59:58.592  6211  6275 E MediaMetadataRetrieverJNI: getFrameAtTime: videoFrame is a NULL pointer

```

Nexolight, 2017-02-13:
Just some more logs:

```
02-13 19:14:45.880  3104  5709 I MediaCodecSource: encoder (video) stopping
02-13 19:14:45.881  3104  5744 E MPEG4Writer: The number of recorded samples is 0
02-13 19:14:45.881  3104  5744 W MPEG4Writer: 0-duration samples found: 1
02-13 19:14:45.881  3104  5744 W MPEG4Writer: 0-duration samples found: 1
02-13 19:14:45.881  3104  5744 I MPEG4Writer: Received total/0-length (0/0) buffers and encoded 0 frames. - Video
02-13 19:14:45.883  3018  5759 I         : [Grace_0.0.5.0][smartAf]AFGetDeviceInfo: illegal af_get_iso, ae_iso 0
02-13 19:14:45.883  3018  5759 I         : [Grace_0.0.5.0][smartAf]AFGetInfo: AFGetDeviceInfo error!
02-13 19:14:45.883  3018  5759 I         : [Grace_0.0.5.0][smartAf]af_return_af_result: mResult[0] = 7, bDarkScreenFlag = 0
02-13 19:14:45.883  3018  5759 I         : [Grace_0.0.5.0][smartAf]af_return_af_result: mResult[0] = 7, bDarkScreenFlag = 0
02-13 19:14:45.888  3515  3863 I vol.Events: writeEvent mute_changed STREAM_RING false
02-13 19:14:45.890  3515  3863 I vol.Events: writeEvent zen_mode_changed off
02-13 19:14:45.890  3515  3863 I vol.Events: writeEvent mute_changed STREAM_SYSTEM false
02-13 19:14:45.891  3104  5709 I MediaCodecSource: encoder (video) stopped
02-13 19:14:45.891  3104  3783 I MediaCodecSource: puller (video) stopping
02-13 19:14:45.892  3104  3783 D CameraSource: reset: E
02-13 19:14:45.892  3104  3783 I Camera2Client: stopRecording: E
02-13 19:14:45.892  3104  3783 E MediaPlayer: Attempt to perform seekTo in wrong state: mPlayer=0xf620a540, mCurrentState=0
02-13 19:14:45.892  3104  3783 E MediaPlayer: start called in state 0
02-13 19:14:45.892  3104  3783 I Camera2Client: stopRecording: hwVideoFps = 0
02-13 19:14:45.892  3104  3783 I Camera2-StreamingProcessor: startStream:Camera 0: type = 2
02-13 19:14:45.893  3104  3783 D CameraSource: reset: X
02-13 19:14:45.893  3104  3783 I MediaCodecSource: puller (video) stopped
02-13 19:14:45.893  3104  3783 D MPEG4Writer: Video track source stopped
02-13 19:14:45.893  3104  3783 D MPEG4Writer: Video track stopped
02-13 19:14:45.893  3104  3783 D MPEG4Writer: Audio track stopping
02-13 19:14:45.893  3104  3783 D MPEG4Writer: Audio track source stopping
02-13 19:14:45.894  3104  5709 I MediaCodecSource: encoder (audio) stopping
02-13 19:14:45.894  3104  5758 I MPEG4Writer: Received total/0-length (12/0) buffers and encoded 11 frames. - Audio
02-13 19:14:45.894  3104  5758 I MPEG4Writer: Audio track drift time: 0 us
02-13 19:14:45.897  5738  5738 I WebViewFactory: Loading com.android.webview version 51.0.2704.106 (code 1)
02-13 19:14:45.898  3515  3863 I vol.Events: writeEvent internal_ringer_mode_changed normal
02-13 19:14:45.899  3104  5709 I MediaCodecSource: encoder (audio) stopped
02-13 19:14:45.900  3104  3783 I MediaCodecSource: puller (audio) stopping
02-13 19:14:45.912  3018  5759 I         : [Grace_0.0.5.0][smartAf]af_doAF: flow: 0, giNumFlagDoAF: 1, afType: 1
02-13 19:14:45.912  3018  5759 D AFDeviceImpl: [90af8a5_D]enter af_get_vsync_time()
02-13 19:14:45.912  3104  5717 I audio_hw_primary: do_in_standby standby: 0, in_device: 128
02-13 19:14:45.912  3018  5759 I         : [Grace_0.0.5.0][smartAf]AFGetDeviceInfo: illegal af_get_iso, ae_iso 0
02-13 19:14:45.912  3018  5759 I         : [Grace_0.0.5.0][smartAf]AFGetInfo: AFGetDeviceInfo error!
02-13 19:14:45.912  3018  5759 I         : [Grace_0.0.5.0][smartAf]af_return_af_result: mResult[0] = 7, bDarkScreenFlag = 0
02-13 19:14:45.912  3018  5759 I         : [Grace_0.0.5.0][smartAf]af_return_af_result: mResult[0] = 7, bDarkScreenFlag = 0
02-13 19:14:45.912  3018  5760 D DeviceImpl: [HWA_CAM3]coff->exposure = 0 ,coff->exposure_time = 0,coff->gain = 0,coff->iso = 0
02-13 19:14:45.913  3018  5760 D DeviceImpl: [HWA_CAM3]virtual int hw::DeviceImpl::getAeCoff(android::HwaIppFrame*, hwa_ae_coff*)() coff->gain_max = 128 ,coff->gain = 0,coff->expo
sure = 0,coff->luma_target_high = 13312,coff->luma_target_low = 11572
02-13 19:14:45.932  3104  5717 I audio_hw_primary: select_devices++ mode[0]
02-13 19:14:45.932  3104  5717 I audio_hw_primary: output_device_id is 2, output_route is media-speaker;input_source_id is 0, input_route is null.
02-13 19:14:45.932  3104  5717 V audio_hw_primary: original selected output route media-speaker input route none
02-13 19:14:45.932  3104  5717 D audio_custom: dev_get_input_path:
02-13 19:14:45.932  3104  5717 V audio_hw_primary: custom selected input route none
02-13 19:14:45.932  3104  5717 D audio_custom: dev_get_output_path:
02-13 19:14:45.932  3104  5717 V audio_hw_primary: custom selected output route media-speaker
02-13 19:14:45.932  3104  5717 V audio_hw_primary: select_devices() devices 0x2 input src 0 output route media-speaker input route none
02-13 19:14:45.932  3104  5717 V audio_hw_primary: forced_out 0 out_device 0x2
02-13 19:14:45.932  3104  5717 I dsp_common: dsp_common_control_algo:set device = 2
02-13 19:14:45.932  3104  5717 I dsp_maxim_ctl: get_smartpa_sence:device = 0x2
02-13 19:14:45.932  3104  5717 I audio_route: route change AUX PGA GAIN value[0] 0
02-13 19:14:45.932  3104  5717 I audio_route: route change MAIN PGA GAIN value[0] 0
02-13 19:14:45.933  3104  5717 I audio_pop: pop change S1 OL SWITCH SWITCH dir 0
02-13 19:14:45.933  3104  5717 I audio_pop: pop change S1 OR SWITCH SWITCH dir 0
02-13 19:14:45.934  3104  5717 I audio_route: route change ADC0R MUX value 1
02-13 19:14:45.934  3104  5717 I audio_route: route change ADC0L MUX value 0
[... cut ...]
02-13 19:14:45.942  5507  5507 E CAM_VideoModule: 	at android.app.ActivityThread.main(ActivityThread.java:5461)
02-13 19:14:45.942  5507  5507 E CAM_VideoModule: 	at java.lang.reflect.Method.invoke(Native Method)
02-13 19:14:45.942  5507  5507 E CAM_VideoModule: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
02-13 19:14:45.942  5507  5507 E CAM_VideoModule: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
02-13 19:14:45.942  5507  5507 V CAM_VideoModule: Deleting video /storage/emulated/0/DCIM/Camera/VID_20170213_191444.mp4.tmp

```

SpeedoWBT, 2017-03-02:
Hi, maybe this could help for next build 👍 
https://github.com/Meticulus/android_device_huawei_hi6250/commit/6ddc966c462cdc3d45d81335c265f250c1de98b6

Nexolight, 2017-03-21:
The issue wasn't yet updated. WIth the hwc from XePeleato and thus the cm libgui library and surfaceflinger  video recording does work. At least the screen recorder and the camera with some but not all apps.

Nexolight, 2017-03-27:
7ceecb8585afa1896b175da2be5b1d54dd13496e fixes a lot of issues
Also video recording which I can confirm does work with footej and probably any app that uses an older api.