CM-Huawei-P8-Development/device_gra_l09 #13: MAC Address is always different after reboot

skrypious, 2017-02-04:
My MAC Address in Cyanogen 13.0 is always different after reboot, whereby I can't connect to WiFi...

skrypious, 2017-02-04:
Ok, 
I found the solution.

In data/misc/wifi

*wpa_supplicant.conf
```
> ctrl_interface=/data/misc/wifi/sockets
> disable_scan_offload=1
> driver_param=use_p2p_group_interface=1p2p_device=1
> update_config=1
> device_name=GRA-L09
> manufacturer=HUAWEI
> model_name=HUAWEI GRA-L09
> model_number=HUAWEI GRA-L09
> serial_number=S7M0215B07021721
> wapi_type=hisi
> device_type=10-0050F204-5
> config_methods=physical_display virtual_push_button
> pmf=1
```

*p2p_supplicant.config
```
> ctrl_interface=/data/misc/wifi/sockets
> driver_param=use_p2p_group_interface=1p2p_device=1
> update_config=1
> device_name=HUAWEI P8
> wapi_type=hisi
> device_type=10-0050F204-5
> config_methods=virtual_push_button physical_display keypad
> p2p_ssid_postfix=-HUAWEI P8
> persistent_reconnect=1
> pmf=1
```

*Add new file with name "macwifi" and type your MAC Address inside this file. Save and reboot

Now Mac Address is static

Nexolight, 2017-02-06:
Might help but it isn't really a fix. More like a workaround.
I can't do that for the rom since the MAC address is supposed to be unique. 

However this file 
`bin/mac_addr_normalization``
was inside the stock rom. It might be a tool to set MAC address to what it is supposed to be.

Nexolight, 2017-05-18:
This is probably fixed with the latest build but I have not yet tested it.

skrypious, 2017-05-30:
MAC Address works properly for me :)

Nexolight, 2017-05-30:
thx for feedback

closed 2017-05-30 by Nexolight