CM-Huawei-P8-Development/device_gra_l09 #14: Youtube hanging on 720p and higher

rubikscraft, 2017-02-10:
I've been testing the rom for some time now,
and discovered that if you watch **some** youtube video's in 720p and higher
the image will freeze but the sound will keep playing.

I have no idea what it is.

Nexolight, 2017-02-13:
I can watch 1080p videos without any issues.
Eighter it's related to the latest memory issues or it's just with some videos and not all. I know what you're talking about but for me even before I did the latest builds that was rarely the case.

skrypious, 2017-02-14:
I can watch 1080p videos too, but videos with 50/60FPS didn't work :/

rubikscraft, 2017-02-15:
You're right it has something to do with the 50/60 hertz videos.

Nexolight, 2017-02-21:
I will investigate in this by time.