CM-Huawei-P8-Development/device_gra_l09 #15: SDCard format broken

Nexolight, 2017-02-13:
Formating the SDCard isn't possible and throws a toast error with a command not found message 
fwe

SpeedoWBT, 2017-02-20:
Hi , maybe look at this commit (Honor 8 by Surdu Pedru)
https://github.com/Huawei-Dev/android_device_huawei_frd/commit/5cf3c1419755c9b1d57b005dcec4d26726b5c1ca

Nexolight, 2017-02-23:
Some of the changes were already done by Paul Fasola.
Furthermore the comment from Meticulus may provide the solution. Just had no time yet.
https://github.com/CM-Huawei-P8-Development/device_gra_l09/commit/3c276bb241a5e56ca4aebf8cd296760b74e6a455

Nexolight, 2017-03-27:
This is the stacktrace from the toast error:
```
03-27 22:32:44.679  4405  4405 E StorageSettings: Failed to partition
03-27 22:32:44.679  4405  4405 E StorageSettings: java.lang.IllegalStateException: command '16 volume partition disk:179_160 private' failed with '400 16 Command failed'
03-27 22:32:44.679  4405  4405 E StorageSettings:       at android.os.Parcel.readException(Parcel.java:1628)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at android.os.Parcel.readException(Parcel.java:1573)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at android.os.storage.IMountService$Stub$Proxy.partitionPrivate(IMountService.java:1067)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at android.os.storage.StorageManager.partitionPrivate(StorageManager.java:706)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at com.android.settings.deviceinfo.StorageWizardFormatProgress$PartitionTask.doInBackground(StorageWizardFormatProgress.jav
a:97)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at com.android.settings.deviceinfo.StorageWizardFormatProgress$PartitionTask.doInBackground(StorageWizardFormatProgress.jav
a:92)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at android.os.AsyncTask$2.call(AsyncTask.java:295)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
03-27 22:32:44.679  4405  4405 E StorageSettings:       at java.lang.Thread.run(Thread.java:818)
```

Nexolight, 2017-03-27:
fixed c1675370e1c53fd1d3b3c54090597f7a7b20d776

closed 2017-03-27 by Nexolight