CM-Huawei-P8-Development/device_gra_l09 #25: Serious app crashes when using heavy apps.

Andsoeg, 2017-04-07:
Feks the game Mushroom Wars space.
If you dont have power connected when playing audio Fx crashes every 5 secound. But when it is connected the game makes the screen look like those old TVs with bo signal. Then phone shuts down. I believe the system gets overloaded.

Nexolight, 2017-04-12:
I will try it

Andsoeg, 2017-04-13:
connected to a power supply, it acts differently. I had to roll back to stock ROM due to these crashes. The phone gets so hot it almost hurts and suddenly the screen gets all black and white, (or the screen loops the previous 2-3 frames) Sometimes it works to shut the screen down and start it again. Also when I used these heavy apps the phone drained almost 1 percent every minute. If I charged it still went down. As you have mentioned, this can have somthing to do with what build I had before I flashed. (B393)

I dont know how i can exctract logs etc so i cannot help you on that part.

Nexolight, 2017-04-13:
As far as I remember I haven't done a lot of gaming on this device. But some low-level GBA emulators, and Antutu are fine. I might also try PPSSPP.

The probably most intense load was caused by a VR app. Full wireless bandwidth, usage of all sensors and h246 decoding at 1080p. It got quite hot but never stoped working. It just dimmed the screen a bit. I will try to reproduce it.

At least I can confirm that audioFX crashes. But so far that's the only case so I probably won't do anything about that as long as not a whole bunch of apps are affected. When charging and not under heavy load it becomes a bit warm but that's all.

Nexolight, 2017-05-18:
So with the latest build audio fx doesn't crash anymore. I can play that game without issues no matter if connected to power or not.

I haven't noticed much warming up from that game.

closed 2017-05-18 by Nexolight

Andsoeg, 2017-06-15:
I have to mention that these flickering is worse than ever before at the moment. 
How can i het the logs when this occurs? Any app?