CM-Huawei-P8-Development/device_gra_l09 #26: double-tap to wake is invalid and 4G network unstable

hilent, 2017-05-27:
1.double-tap to wake is invalid 
2.4G network is unstable, sometimes it works normally, but sometimes it cannot link to internet at all, and this situation is pretty frequent

the model of my device is gra-ul00, the rom before is b370 ,andorid 6.0

thx!

Nexolight, 2017-05-30:
Well the ROM is for the GRA-L09 and other devices might work but are not tested and there's no guarantee for anything to work.

2.4G Wifi was never reported on the L09 so far and double tap to wake is not implemented. This isn't a bug per definition.

It might be added one day

hilent, 2017-05-30:
oh, i'm sorry but i have to explain that, "4G network" refer to Mobile Data(lte-fdd) rather than 2.4G wifi. wifi works normally entirely, but Mobile Data will be disconnected now and again.
2g network works perfectly, phonecall and messages are perfectly normal.

Nexolight, 2017-05-30:
Hm I never had any disconnects under non-special circumstances where it would be normal nor was it reported ao far.

Logs when it happens might help but apart from that I can only guess.

Try to change /system/build.props ro.telephony.default_network from 9 to 10 and reboot.

When that doesn't help maybe it could help to modify some of the lines below the #modem related? comment. I just can't tell what exactly as most of these values come from the stock rom.

the getprop command at runtime from a stock ul00 would deliver the correct values for sure.

hilent, 2017-05-31:
here is bulid.props for ul00, i extract from emui4.0 b370 for ul00 /system.

https://mega.nz/#!ReAznLpA!OmRhWUCO_cvMDGmzX5C3LBIytytQ6NMKscvGNk3CKes

ahout the "ro.telephony.default_network", the bulid.props of emui donot have this item, i'll try to install cm13 again and try to change this one to see if it works, it maybe require 2~3days, because i have go back to emui for this issue.

Nexolight, 2017-05-31:
It's not just that file. Some of the binaries set props as well. And the props for the modem are enumerated with a huge list consisting out of blocks for each modem from huawei. modem/modem.props or smth. like that. I'm mobile, can't look up the exact name right now.

With getprop at *runtime* you would get all of them.

sibajie, 2017-06-17:
我是中国人，我是用移动定制版TL-00破解双4G，变成UL-00，然后刷入cm13变为GRA-L09，此刻双4G正常，然后刷入双卡的包，系统提示sim卡未插入，wipe尝试了也没有效果。

hilent, 2017-06-17:
@sibajie 你好，我也是先破解成了双4G版，从ul00的3.1一路升级到4.0.2，然后再刷的CM13，开机后都正常，但是会出现我说的4G断流，移动网络时有时无，没出现未插入SIM的情况
另外请问，“然后刷入双卡的包”中“双卡的包”是什么？

closed 2017-06-29 by hilent