CM-Huawei-P8-Development/device_gra_l09 #29: I have some trouble with Compile. Please help me ! Thank you!

him1234, 2017-07-31:
android@androidpc:~/lineageos$ . build/envsetup.sh
including device/generic/mini-emulator-arm64/vendorsetup.sh
including device/generic/mini-emulator-armv7-a-neon/vendorsetup.sh
including device/generic/mini-emulator-x86_64/vendorsetup.sh
including device/generic/mini-emulator-x86/vendorsetup.sh
including device/htc/hiaeul/vendorsetup.sh
including device/huawei/hwgra/vendorsetup.sh
including vendor/cm/vendorsetup.sh
including sdk/bash_completion/adb.bash
including vendor/cm/bash_completion/git.bash
including vendor/cm/bash_completion/repo.bash
android@androidpc:~/lineageos$ breakfast hwgra
including vendor/cm/vendorsetup.sh
Trying dependencies-only mode on a non-existing device tree?

============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=7.1.2
LINEAGE_VERSION=14.1-20170731-UNOFFICIAL-hwgra
TARGET_PRODUCT=lineage_hwgra
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a53
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.10.0-28-generic-x86_64-with-Ubuntu-17.04-zesty
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47D
OUT_DIR=/home/android/lineageos/out
RECOVERY_VARIANT=twrp
============================================
android@androidpc:~/lineageos$ make
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=7.1.2
LINEAGE_VERSION=14.1-20170731-UNOFFICIAL-hwgra
TARGET_PRODUCT=lineage_hwgra
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a53
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.10.0-28-generic-x86_64-with-Ubuntu-17.04-zesty
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47D
OUT_DIR=/home/android/lineageos/out
RECOVERY_VARIANT=twrp
============================================
Running kati to generate build-lineage_hwgra.ninja...
/home/android/lineageos/out/build-lineage_hwgra.ninja is missing, regenerating...
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=7.1.2
LINEAGE_VERSION=14.1-20170731-UNOFFICIAL-hwgra
TARGET_PRODUCT=lineage_hwgra
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a53
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.10.0-28-generic-x86_64-with-Ubuntu-17.04-zesty
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47D
OUT_DIR=/home/android/lineageos/out
RECOVERY_VARIANT=twrp
============================================
build/core/binary.mk:1253: external/deqp/Android.mk: libdeqp: Unused source files: \
build/core/binary.mk:1253: external/deqp/Android.mk: libdeqp: Unused source files: \
build/core/binary.mk:1253: external/pdfium/third_party/pdfiumbigint.mk: libpdfiumbigint: Unused source files: bigint/BigInteger.cc bigint/BigIntegerUtils.cc bigint/BigUnsigned.cc bigint/BigUnsignedInABase.cc
build/core/binary.mk:1253: external/pdfium/third_party/pdfiumbigint.mk: libpdfiumbigint: Unused source files: bigint/BigInteger.cc bigint/BigIntegerUtils.cc bigint/BigUnsigned.cc bigint/BigUnsignedInABase.cc
./external/speex/Android.mk:56: TODOArm64: enable neon in libspeex
build/core/package_internal.mk:143: Empty argument supplied to find-subdir-assets
find: ‘/home/android/lineageos/out/target/common/obj/SHARED_LIBRARIES/libwifi-hal-mock_intermediates’: There is no file or directory
build/core/package_internal.mk:143: Empty argument supplied to find-subdir-assets
build/core/shared_library.mk:12: *** audio.hdmi.hi3635: LOCAL_MODULE_PATH for shared libraries is unsupported in multiarch builds, use LOCAL_MODULE_RELATIVE_PATH instead.
build/core/ninja.mk:166: recipe for target '/home/android/lineageos/out/build-lineage_hwgra.ninja' failed
make: *** [/home/android/lineageos/out/build-lineage_hwgra.ninja] Error 1

#### make failed to build some targets (45 seconds) ####

-----------------------------------------------------------------------------------------------------------------------------------
This is the compilation log.

How can I solve such the problem?

him1234, 2017-07-31:
and this log
---------------------------------------------------------------------------
sem_open.c:333：警告： the use of `mktemp' is dangerous, better use `mkstemp' or `mkdtemp'
Running kati to generate build-lineage_hwgra.ninja...
/home/android/lineageos/out/build-lineage_hwgra.ninja is missing, regenerating...
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=7.1.2
LINEAGE_VERSION=14.1-20170731-UNOFFICIAL-hwgra
TARGET_PRODUCT=lineage_hwgra
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a53
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.10.0-28-generic-x86_64-with-Ubuntu-17.04-zesty
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47D
OUT_DIR=/home/android/lineageos/out
RECOVERY_VARIANT=twrp
============================================
Checking build tools versions...
build/core/binary.mk:1253: external/deqp/Android.mk: libdeqp: Unused source files: \
build/core/binary.mk:1253: external/deqp/Android.mk: libdeqp: Unused source files: \
build/core/binary.mk:1253: external/pdfium/third_party/pdfiumbigint.mk: libpdfiumbigint: Unused source files: bigint/BigInteger.cc bigint/BigIntegerUtils.cc bigint/BigUnsigned.cc bigint/BigUnsignedInABase.cc
build/core/binary.mk:1253: external/pdfium/third_party/pdfiumbigint.mk: libpdfiumbigint: Unused source files: bigint/BigInteger.cc bigint/BigIntegerUtils.cc bigint/BigUnsigned.cc bigint/BigUnsignedInABase.cc
./external/speex/Android.mk:56: TODOArm64: enable neon in libspeex
build/core/package_internal.mk:143: Empty argument supplied to find-subdir-assets
find: ‘/home/android/lineageos/out/target/common/obj/SHARED_LIBRARIES/libwifi-hal-mock_intermediates’: 没有那个文件或目录
build/core/package_internal.mk:143: Empty argument supplied to find-subdir-assets
build/core/shared_library.mk:12: *** audio.hdmi.hi3635: LOCAL_MODULE_PATH for shared libraries is unsupported in multiarch builds, use LOCAL_MODULE_RELATIVE_PATH instead.
build/core/ninja.mk:166: recipe for target '/home/android/lineageos/out/build-lineage_hwgra.ninja' failed
make: *** [/home/android/lineageos/out/build-lineage_hwgra.ninja] Error 1

#### make failed to build some targets (14 seconds) ####

android@androidpc:~/lineageos$

Nexolight, 2017-08-07:
Could you please put the logs in code format it's huge o.o
This is lineage so I'm not really familiar with the pitfalls. AD2076 made the lineage branch.

Have you checked out the lineage branch [here](https://github.com/CM-Huawei-P8-Development/local_manifests_gra_l09) ? The default branch in the manifest is set to CM which points to the CM sources and not the lineage one. 

All I can see is that it fails to make the audio.hdmi.hi3635 module probably because the .mk file there uses LOCAL_MODULE_PATH instead of LOCAL_MODULE_RELATIVE_PATH. That means it wasn't made for multiple architectures as stated in the log. 

When I look at the environment variables however I think they are wrong.

This is how I build the CM rom. It's the same for Lineage most likely.
cd to your ndk dir and then (with python 2.7 as default)

`source build/envsetup`
`lunch cm_hwgra-userdebug`
`make clobber` (optional - cleanup)
`mka`

The lunch command should output smth like this (with ofc different values):

```
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=6.0.1
CM_VERSION=13.0-20170807-UNOFFICIAL-hwgra
TARGET_PRODUCT=cm_hwgra
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a53
HOST_ARCH=x86_64
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.10.13-1-ARCH-x86_64-with-glibc2.2.5
HOST_BUILD_TYPE=release
BUILD_ID=MOB31K
OUT_DIR=/home/nexolight/ndk/out
============================================
```

him1234, 2017-08-16:
Thank you very much！！！！！
but,I compiled Android 7.0  Lineageos 14.1
Can you help me？@Nexolight

Nexolight, 2017-08-28:
I can't help with that apart from what I wrote which is most likely still valid.

AD2076 might be able to help with that as he's doing the port for Android 7.0.
I'm however not sure how far he got until now.

closed 2017-10-29 by him1234