CM-Huawei-P8-Development/device_gra_l09 #31: Can I use the Makefiles to compile Lineageos 14.1?

him1234, 2017-10-29:
How can I compile with lineageos 14.1 correctly? and @AD2076 will compile the Android 7.0 to HUAWEI p8 in the future?

AD2076, 2017-11-05:
It could compile but it probably won't even boot

him1234, 2017-11-10:
`android@android-MS-7918:~/los$ brunch hwgra
including vendor/cm/vendorsetup.sh
Trying dependencies-only mode on a non-existing device tree?

============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=7.1.2
LINEAGE_VERSION=14.1-20171110_142515-UNOFFICIAL-hwgra
TARGET_PRODUCT=lineage_hwgra
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a53
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.10.0-38-generic-x86_64-with-Ubuntu-16.04-xenial
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47F
OUT_DIR=/home/android/los/out
RECOVERY_VARIANT=twrp
============================================
make: Entering directory '/home/android/los'
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=7.1.2
LINEAGE_VERSION=14.1-20171110_142515-UNOFFICIAL-hwgra
TARGET_PRODUCT=lineage_hwgra
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a53
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.10.0-38-generic-x86_64-with-Ubuntu-16.04-xenial
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47F
OUT_DIR=/home/android/los/out
RECOVERY_VARIANT=twrp
============================================
Running kati to generate build-lineage_hwgra.ninja...
/home/android/los/out/build-lineage_hwgra.ninja is missing, regenerating...
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=7.1.2
LINEAGE_VERSION=14.1-20171110_142515-UNOFFICIAL-hwgra
TARGET_PRODUCT=lineage_hwgra
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a53
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.10.0-38-generic-x86_64-with-Ubuntu-16.04-xenial
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47F
OUT_DIR=/home/android/los/out
RECOVERY_VARIANT=twrp
============================================
/bin/sh: 1: cd: can't cd to /home/android/los/out/target/product/hwgra/obj/busybox
Makefile:112: output directory "/home/android/los/out/target/product/hwgra/obj/busybox" does not exist
Makefile:112: *** On AOSP repo, type 'mma' to build or set O=out/<folder> 。 停止。
/bin/sh: 1: cd: can't cd to /home/android/los/out/target/product/hwgra/obj/busybox
Makefile:112: output directory "/home/android/los/out/target/product/hwgra/obj/busybox" does not exist
Makefile:112: *** On AOSP repo, type 'mma' to build or set O=out/<folder> 。 停止。
/bin/sh: 1: cd: can't cd to /home/android/los/out/target/product/hwgra/obj/busybox
Makefile:112: output directory "/home/android/los/out/target/product/hwgra/obj/busybox" does not exist
Makefile:112: *** On AOSP repo, type 'mma' to build or set O=out/<folder> 。 停止。
build/core/shared_library.mk:12: *** audio.hdmi.hi3635: LOCAL_MODULE_PATH for shared libraries is unsupported in multiarch builds, use LOCAL_MODULE_RELATIVE_PATH instead.
build/core/ninja.mk:166: recipe for target '/home/android/los/out/build-lineage_hwgra.ninja' failed
make: *** [/home/android/los/out/build-lineage_hwgra.ninja] Error 1
make: Leaving directory '/home/android/los'

#### make failed to build some targets (8 seconds) ####
`

him1234, 2017-11-10:
```

> android@android-MS-7918:~/los$ brunch hwgra
> including vendor/cm/vendorsetup.sh
> Trying dependencies-only mode on a non-existing device tree?
> 
> ============================================
> PLATFORM_VERSION_CODENAME=REL
> PLATFORM_VERSION=7.1.2
> LINEAGE_VERSION=14.1-20171110_142515-UNOFFICIAL-hwgra
> TARGET_PRODUCT=lineage_hwgra
> TARGET_BUILD_VARIANT=userdebug
> TARGET_BUILD_TYPE=release
> TARGET_BUILD_APPS=
> TARGET_ARCH=arm64
> TARGET_ARCH_VARIANT=armv8-a
> TARGET_CPU_VARIANT=cortex-a53
> TARGET_2ND_ARCH=arm
> TARGET_2ND_ARCH_VARIANT=armv7-a-neon
> TARGET_2ND_CPU_VARIANT=cortex-a53
> HOST_ARCH=x86_64
> HOST_2ND_ARCH=x86
> HOST_OS=linux
> HOST_OS_EXTRA=Linux-4.10.0-38-generic-x86_64-with-Ubuntu-16.04-xenial
> HOST_CROSS_OS=windows
> HOST_CROSS_ARCH=x86
> HOST_CROSS_2ND_ARCH=x86_64
> HOST_BUILD_TYPE=release
> BUILD_ID=NJH47F
> OUT_DIR=/home/android/los/out
> RECOVERY_VARIANT=twrp
> ============================================
> make: Entering directory '/home/android/los'
> ============================================
> PLATFORM_VERSION_CODENAME=REL
> PLATFORM_VERSION=7.1.2
> LINEAGE_VERSION=14.1-20171110_142515-UNOFFICIAL-hwgra
> TARGET_PRODUCT=lineage_hwgra
> TARGET_BUILD_VARIANT=userdebug
> TARGET_BUILD_TYPE=release
> TARGET_BUILD_APPS=
> TARGET_ARCH=arm64
> TARGET_ARCH_VARIANT=armv8-a
> TARGET_CPU_VARIANT=cortex-a53
> TARGET_2ND_ARCH=arm
> TARGET_2ND_ARCH_VARIANT=armv7-a-neon
> TARGET_2ND_CPU_VARIANT=cortex-a53
> HOST_ARCH=x86_64
> HOST_2ND_ARCH=x86
> HOST_OS=linux
> HOST_OS_EXTRA=Linux-4.10.0-38-generic-x86_64-with-Ubuntu-16.04-xenial
> HOST_CROSS_OS=windows
> HOST_CROSS_ARCH=x86
> HOST_CROSS_2ND_ARCH=x86_64
> HOST_BUILD_TYPE=release
> BUILD_ID=NJH47F
> OUT_DIR=/home/android/los/out
> RECOVERY_VARIANT=twrp
> ============================================
> Running kati to generate build-lineage_hwgra.ninja...
> /home/android/los/out/build-lineage_hwgra.ninja is missing, regenerating...
> ============================================
> PLATFORM_VERSION_CODENAME=REL
> PLATFORM_VERSION=7.1.2
> LINEAGE_VERSION=14.1-20171110_142515-UNOFFICIAL-hwgra
> TARGET_PRODUCT=lineage_hwgra
> TARGET_BUILD_VARIANT=userdebug
> TARGET_BUILD_TYPE=release
> TARGET_BUILD_APPS=
> TARGET_ARCH=arm64
> TARGET_ARCH_VARIANT=armv8-a
> TARGET_CPU_VARIANT=cortex-a53
> TARGET_2ND_ARCH=arm
> TARGET_2ND_ARCH_VARIANT=armv7-a-neon
> TARGET_2ND_CPU_VARIANT=cortex-a53
> HOST_ARCH=x86_64
> HOST_2ND_ARCH=x86
> HOST_OS=linux
> HOST_OS_EXTRA=Linux-4.10.0-38-generic-x86_64-with-Ubuntu-16.04-xenial
> HOST_CROSS_OS=windows
> HOST_CROSS_ARCH=x86
> HOST_CROSS_2ND_ARCH=x86_64
> HOST_BUILD_TYPE=release
> BUILD_ID=NJH47F
> OUT_DIR=/home/android/los/out
> RECOVERY_VARIANT=twrp
> ============================================
> /bin/sh: 1: cd: can't cd to /home/android/los/out/target/product/hwgra/obj/busybox
> Makefile:112: output directory "/home/android/los/out/target/product/hwgra/obj/busybox" does not exist
> Makefile:112: *** On AOSP repo, type 'mma' to build or set O=out/<folder> 。 停止。
> /bin/sh: 1: cd: can't cd to /home/android/los/out/target/product/hwgra/obj/busybox
> Makefile:112: output directory "/home/android/los/out/target/product/hwgra/obj/busybox" does not exist
> Makefile:112: *** On AOSP repo, type 'mma' to build or set O=out/<folder> 。 停止。
> /bin/sh: 1: cd: can't cd to /home/android/los/out/target/product/hwgra/obj/busybox
> Makefile:112: output directory "/home/android/los/out/target/product/hwgra/obj/busybox" does not exist
> Makefile:112: *** On AOSP repo, type 'mma' to build or set O=out/<folder> 。 停止。
> build/core/shared_library.mk:12: *** audio.hdmi.hi3635: LOCAL_MODULE_PATH for shared libraries is unsupported in multiarch builds, use LOCAL_MODULE_RELATIVE_PATH instead.
> build/core/ninja.mk:166: recipe for target '/home/android/los/out/build-lineage_hwgra.ninja' failed
> make: *** [/home/android/los/out/build-lineage_hwgra.ninja] Error 1
> make: Leaving directory '/home/android/los'
> 
> #### make failed to build some targets (8 seconds) ####

```

him1234, 2017-11-10:
`Now I'm trying to use lineageos14.1, and there are still a lot of mistakes`

Nexolight, 2018-04-18:
I honestly don't know. AD2076 is right. It might compile but it will most likely not boot.

There are multiple issues which one can't workaround:
1. Android 7 requires a more recent graphic chip
2. The libraries used from the blob files were compiled with the android 6 version from huawei. The source meanwhile moved on and there's a good chance that these won't be compatible anymore, causing a shitton of segfaults.
3. Same goes for the kernel. I needed to patch some things in order to use it with Android 6.  I'm afraid it will become even a bigger problem with Android 7. One could just put in the lastest kernel but then good luck making the huawei kernelspace drivers compile

closed 2018-07-14 by him1234