CM-Huawei-P8-Development/device_gra_l09 #34: No wifi tethering [again]

Beheavyon, 2018-03-05:
Hello.
This functions is should be working (as you said).
But i can manipulate with icon what disapper after ending the editing. Also aps from market (i tested many of it) doesn't work. Other devices can not conect to phone.

Nexolight, 2018-04-18:
[image]
[image]
[image]

Nexolight, 2018-04-18:
Settings -> More... -> Tethering & mobile hotspot -> Mobile WLAN-Hotspot.

It is there and works for me.

closed 2018-04-18 by Nexolight